XKNX/xknx #417: Time exposed wrong to bus in some home assistant configurations

hansomatik, 2020-09-22:
**Description of problem:**

using Home-Assistant knx plugin

Problem

In home assistant for knx time expose the system base time is used. In many installation types like hass os / docker based the system time is wrong / differs from frond end time. This isn't easy to change because of unwanted ssh access to base image.
Many users seem to have that problem:
https://community.home-assistant.io/t/time-is-exposing-wrongly-to-knx-bus/115877

Possible solution:
Is it possible to change the time used to the home assistant frontend time?
I know its not directly xknxs fault. But its the by far easiest way to get a working solution by doing an option on the xknx side.

farmio, 2020-09-22:
Isn't "System time is wrong" the issue here 😂
In docker you can just pass the system time to the container with a read-only volume `/etc/localtime:/etc/localtime:ro`

marvin-w, 2020-09-22:
Or just expose the env variable `TZ=Europe/Berlin` in the docker container and it should work.

But I have to agree with @farmio here. This is an issue with your operating system settings (be it docker or standalone).