XKNX/xknx #420: Request: support DPT1.007 (1bit) setpoint shift mode for climate component

da-anda, 2020-09-23:
Hey, my MDT climate actuators support a "simplified" setpoint shift mode using DPT1.007 which allows to specify the desired step size centrally in the actuator and control units simply send 0 to reduce the temp one step or 1 to increase it one step. This IMO greatly simplifies the configuration if multiple control units (switches/visu) are used to control the room temperature and you want a unified behavior across all.

Would it be possible to add support for DPT1.007 as `setpoint_shift_mode` to the climate component? Thanks.

da-anda, 2020-09-25:
here the manual of my actuator if it is of any help (sorry, German only, but AFAIK you have some German team members)
Pages 46 and 48 https://www.mdt.de/download/MDT_THB_Heizungsaktor_02.pdf

farmio, 2020-09-25:
MDT is the only manufacturer using this Schema for setting temperatures, as far as I'm concerned. 
Their actuators also support more widely used options like DPT 6.010

da-anda, 2020-09-25:
so in order to use temperature controls from both, KNX switches and HA I have to reprogram everything to use DPT6.010 and duplicate the step size config everywhere - nice :(
(well, or write my own KNX integration)

closed 2020-09-25 by da-anda

farmio, 2020-09-25:
Or use `knx.send` service.

da-anda, 2020-09-25:
I will either have to write a custom lovelace component to use `knx.send` or a custom HA integration (and then be able to use any lovelace component). So unnecessary overhead either way.

da-anda, 2020-09-26:
so I was trying a DPT 6.010 config for one room, and it somewhat works, but it's super annoying having to configure min/max and step size for each and every component/device dealing with temperature changes. Who came up with this concept over at KNX - super annoying and highly error prone. DPT1.007 at least to me would make much more sense as you end up with one central config. But well, guess I have to live with it.

farmio, 2020-09-26:
Don't you need limits and stepsize for 1bit too? With DPT 6/9 you don't spam the bus if you want to make changes > stepsize. It's always just 1 telegram.

da-anda, 2020-09-26:
for 1bit, I set the limit and step size once in the heating actuator, and every "control unit" (like my wall switches or HA) only have to say temp+/- and that's it. They do not have to know anything about the min/max temp shift or the step size in order to correctly calculate the offset for display or the message payload, nothing. So if at some point you decide to increase the max temp shift or want a finer/larger grained temp control (different step size) you only change it once in the heating actuator and be done. No need to adjust anything in any of the control units.
But yes, in theory they will also send a message to the bus when the min/max offset is reached. But since you can instantly see there is no change anymore, it's not like you would manually flood the bus.

I think I will try a custom knx component for HA and stick to 1bit since I am no fan at all of having to sync configurations all over the place in case of a tiny change. "All" I have to do is to basically have stubs that import/extend the default knx integration and add a bit of extra logic to the climate component to not forward step +/- to xknx but use knx.send. Well, at least in theory. But since I am no Python dev, know quite little on how HA is working internally and know nothing about xknx, it likely will become a larger task.

farmio, 2020-09-26:
There is no +- from HA climate. You just get a new temperature value that shall be set. So you have to calculate current - new / stepsize to know how many telegrams to send -> you need the stepsize.

da-anda, 2020-09-26:
yeah, just noticed after checking the python components. Whelp, that much for my plan. So probably a custom lovelace component then - at least on the long run (don't like the climate components I tested so far anyways). But that's something for the winter months with hopefully a bit more sparetime.

Thanks for your help and sorry for the noise.

Olum-hack, 2022-02-21:
Hello,
I know that the topic is old but I have exectlythe same.
Did someone already find a solution? Or can it be integrated into the xknx code?

marvin-w, 2022-02-21:
Which actuator do you have?