XKNX/xknx #427: Coveralls adjustment

marvin-w, 2020-09-25:
I'd like to change the behavior of coveralls. Currently, even if the coverage only decreases for just 0,003% the build will fail. We should add a target value for the coverage - 90% is usually a really good value - and only fail the build if we go under that 90%.

marvin-w, 2020-09-25:
@farmio @Julius2342 what do you think?

farmio, 2020-09-25:
Well then one can make untested changes until 90% and then we are back to current. 
Can we make it so it fails the build only if coverage decreases by more than 10 lines / 0,01% ?
I personally think it's ok to fail and we can merge it if we feel save. It's a good reminder to write tests.

marvin-w, 2020-09-25:
Nah then just leave it as it is :).

closed 2020-09-25 by marvin-w

Julius2342, 2020-09-26:
I agree with what @farmio said :-)