XKNX/xknx #435: Feature Request: lock and diagnostic attributes for XKNX devices

da-anda, 2020-09-28:
Hi,

would it be possible to extend light, cover, switch and climate (probably even more/all) entities with GAs for additional attributes to see in which actual state they are? As you know, in KNX most things can be in a locked, have a locked/alarm state and have diagnostic related values you can read out. So my request would be to add the following GAs:

light, cover, switch, climate (all?)
- lock_state (DPT 1.011)
- lock (DPT 1.003) (once cover entities can be locked in HA)
- diagnostic_state (DPT 16.000/16.001 - string)

For covers, it would be also nice to have an attribute GA for
- automatic_shading_state (DPT 1.011)

If you don't want to add these to the XKNX entities, it would be nice to be able to define custom attributes by specifying an address, DPT and name. Something like:
```
knx:
  covers:
    - name: "Home Diningroom"
      move_long_address: '2/1/35'
      move_short_address: '2/2/35'
      position_address: '2/3/35'
      angle_address: '2/4/35'
      position_state_address: '2/5/35'
      angle_state_address: '2/6/35'
      travelling_time_down: 19
      travelling_time_up: 19
      attributes:
        - name: "Automatic shading"
           type: 'bool'
           state_address: '2/7/35'
        - name: "Diagnostic"
           type: 'string'
           state_address: '30/1/35'
```

Thanks.

da-anda, 2020-10-14:
sorry for being impatient. But any comment on this?

- [ ] Good idea?
- [ ] Bad idea?
- [ ] Way to hard to implement with the current internal structure?
- [ ] Matt Damon!

marvin-w, 2020-10-14:
Was the idea to have those available in HA too? If yes, then no we can't do that as HA requires each attribute to be defined within the code. If this is only relevant for XKNX then yes, this might be possible together with the new config schema - but you will *not* see those attributes in HA.

da-anda, 2020-10-14:
yes, my idea was to have those available in HA too (as attributes, or if the KNX component would create config flow devices, then those additional entities could be grouped by the device and exposed as either attribute (if it's just a state) or entity/switch etc).

But if I got you right, there is no way that a HA component provides additional attributes for entities? Doesn't the zwave component do that already?

marvin-w, 2020-10-14:
You can have additional attributes, but those will **all** need to be defined upfront. The user cannot create custom ones for entities.

marvin-w, 2020-10-15:
@da-anda: Idea: We could prepopulate an enum with all allowed attribute types and then use that. Would that be enough for you? Do you have any other attributes except the ones mentioned in the ticket in mind?

da-anda, 2020-10-16:
nah, no need for any hacks just for me. It was just a suggestion to have a way to basically group related GAs into one entity (or `device` with multiple entities if you want to use HA terms).

But if you think those attributes are of general value to everyone, see the first post for my proposal. Since in KNX basically almost everything can be locked or have a diagnostic GA, having those states as additional attributes in HA would be nice. The generic approach was only a suggestion in case you would not like to have a fixed list

a005, 2020-10-27:
It may be introduced in at the Device level not at the Entity level.
HA now has a Device abstraction (not supported by KNX integration) where each Device may enclose one or many Entites.
For example zigbee switch device consists of:
- switch entity (switch type)
- battery state entity (sensor type).

farmio, 2020-10-27:
@a005 how would the HA-device abstraction fit to KNX? A xknx-device (HA-Entity) is not a collection of physical devices nor is it a subset of just one physical device.

a005, 2020-10-27:
This should depend on the user's configuration. Many KNX objects can be represented as Devices  - switches, lights, covers (no matter real or virtual this device is). I have no detailed view right now. But believe knx-virtual-devices will be usefull for Automation UI also  (handling knx_event in triggers it's a pain):
For example, I'd preffer use virtual "Central switch" with configured group address 1/0/0 and in Automation UI could choose:
trigger type: Device
device: Central switch
trigger: "turn off" button pressed.
This is impossible to do with an entity, only with a knx_event handling (no user friendly).

farmio, 2020-10-27:
Wo basically everything that is currently a HA-entity should be a HA-device?

Wouldn't your automation (from the UI) work with
trigger type: state 
entity: switch.central
to: "off"
?

a005, 2020-10-28:
@farmio,
_everything that is currently a HA-entity should be a HA-device?_ I see two options: one Device for each entity or one Device for All entities, but user can able to combine some enities in one device (like cover and lock_state, error_message) and to combine some entities with "things" without state for sending events (like a push button). 

_Wouldn't your automation (from the UI) work with trigger type: state?_ - No. This trigger fires only if state is changed but central switch never has a state 'on'. Generally speaking "Central switch" doesn't have a state, it should be treated like a push button. "Scene switch" doesn't have a state too.
So I need use this for central switch:

```
  trigger:
  #for room central switch
  - event_data:
      address: 1/0/15
    event_type: knx_event
    platform: event
  #for all home central switch
  - event_data:
      address: 1/0/0
    event_type: knx_event
    platform: event
```
and for turn on scene by its number:
```
    trigger:
      - platform: event
        event_type: knx_event
        event_data:
          address: "1/6/3"
    condition: []
    action:
      #look up the scene number in received event and turn on the scene with this number
      - service: script.turn_on_scene_by_number
        data_template:
          area: kitchen
          number: '{{trigger.event.data.data[0]}}'
```

farmio, 2020-10-28:
Yes, HA doesn't really have a pushbutton concept do we need to fake it with resetting state. 
For the automations you have 2 options to avoid knx_event:
- use a switch with `reset_after` - this would send a 1 and then a 0 to the bus. Every 1 from the bus would trigger a state update. (not sure if this is already upstream; here will also be an invert option soon)
- use a binary_sensor with `ignore_internal_state = True` or `reset_after` - whatever fits your needs 

The devices thing would, to my knowledge, require us to use config flow or provide a unique_id. There are already some conversatios about this in other issues.

PR #654 (2021-04-04): Add support for cover locking (weather alarms etc.)

## Description

This PR adds support for a dedicated `movement_blocked` attribute in HA.

[image]

Fixes #435, fixes #72

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

## Checklist:

- [x] I have performed a self-review of my own code
- [x] The documentation has been adjusted accordingly
- [x] The [... cut]
closed 2021-04-04 by marvin-w