XKNX/xknx #440: HA Cover: add stepwise tilt function

farmio, 2020-09-30:
Implement https://github.com/home-assistant/core/blob/e38ae1b188ffac02017c150f416f210f13190799/homeassistant/components/cover/__init__.py#L292
, close and stop tilt for stepwise setting tilt angles in our cover entity. 
This probably requires setting a steptime or even total tilttime to calculate angle (like position in travelcalculator).

crazyfx1, 2021-02-13:
Basically thats already implemented for the KNX side. 
Those 2 methods send the correct values to the bus:
https://github.com/XKNX/xknx/blob/a387710d7727866107215e445babe842a4522fae/xknx/devices/cover.py#L211

If you want to integrate the travelcalculator, you would need to travel the travelcalculator for the steptime as mentioned in your description.

crazyfx1, 2021-02-23:
This was now merged with #602 

I would keep the current xknx implementation and not add anything to the travelcalcator because the tilt position is currently not reflected in the travelcalculator.

closed 2021-02-23 by farmio