XKNX/xknx #450: Exposure of time to KNX bus doesn't seem to work anymore

FredericMa, 2020-10-05:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

The exposure of the time to the KNX bus doesn't seem to work anymore.

**Version information:**
- xknx / Home-Assistant release with the issue: 0.13.0 / 0.115.3
- last working xknx / Home-Assistant release (if known): can't confirm when it broke but I guess since the last xknx upgrade.

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```yaml
knx: 
  routing:
    local_ip: 192.168.50.101
  expose:
  - type: 'time'
    address: '6/0/0'
```
**Traceback (if applicable):**
I enabled debug logging in Home Asisstant for both the KNX and XKNX components but I can't find anything in the logs about either 'time' or the '6/0/0' group address.

marvin-w, 2020-10-05:
Fixed in 0.14.0 - HA 0.116.0

closed 2020-10-05 by marvin-w

FredericMa, 2020-10-05:
Excellent! Thanks!

marvin-w, 2020-10-05:
Beta is already out. Feel free to test it. Otherwise just wait for the release :)