XKNX/xknx #465: Feature request: add the "ignore_internal_state" property to "Sensors" the same way it works already for "Binary Sensors"

kalhimeo, 2020-10-14:
Hi,

In the last release the property "ignore_internal_state" was added to the knx "binary sensor", and it would be great to have this property for knx "sensors" as well.

For example I am using a sensor with type "scene_number" together with an automation in order to execute some actions within HA when a knx scene number is called, but if the same scene is called two times without any other scene number in between, the value of the HA sensor state is not changed and the automation is not called again.

Note : the KNX scene entity is no help in this case as its state in HA does not change when triggered (it stays on "scening") and is only used the other way around (make HA to trigger a KNX scene).

Thank you very much in advance !

**Description of problem:**

- [ ] using xknx standalone
- [ ] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

**Traceback (if applicable):**

farmio, 2020-10-14:
I like the idea. For now as a workaround you can use the HA event fired by the knx integration `knx_event`. For this you will need to configure 'fire_event' and 'fire_event_filter' 
See https://www.home-assistant.io/integrations/knx/ (we should document this better 😶)

kalhimeo, 2020-10-14:
Hi farmio, thank you for the suggestion, that would make the trick as well.
At the moment I added an extra action at the end of my automation in order to trigger a dummy scene number on the KNX bus. The state of the sensor entity is then "reset" to this dummy number and ready to be triggered when the target scene number is activated again.

marvin-w, 2020-10-14:
Thanks! We will proceed here once we have time for it :-D. I'll tag this as Hacktoberfest, maybe someone wants to do it :)

kalhimeo, 2020-10-14:
Thank you marvin :-)

closed 2020-10-17 by marvin-w

kalhimeo, 2020-10-19:
Thank you very much marvin !

PR #469 (2020-10-17): Add always_callback option to Sensor

## Description

Allow sensors to trigger the update callbacks even if the internal state did not change.
This feature is opt in and disabled per default.

Fixes #465

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

## Checklist:

- [x] I have performed a self-review of my own code
- [x] The documentation has been  [... cut]
commit 93f835e3 (2020-10-17): Add always_callback option to Sensor (#469)

Allow sensors to trigger the update callbacks even if the internal state did not change.
This feature is opt in and disabled per default.

Fixes #465