XKNX/xknx #466: Feature request: Add Fan mode and Swing mode to climate KNX devices

adorobis, 2020-10-14:
Hi All,
Would it be possible to add Fan Mode and Swing Mode to KNX Climate devices? It is supported by Home Assistant in other Climate integrations (e.g. Sensibo Climate). If needed I can help with testing, I have a KNX modules for Mitsubishi A/C units which enable to control all features of the A/C unit including Fan modes and Swing modes. If needed I can also share which DPTs are used to control those modes.

marvin-w, 2020-10-14:
Please share as much information as possible :).

adorobis, 2020-10-14:
OK, so here is some more info. I'm using Intesis Box ME-AC-KNX-1-V2 which is dedicated to Mitsubishi A/C units, here I have printed the group objects documentation https://github.com/adorobis/xknx/blob/main/KNX%20AC.pdf (don't know if there is a better way to get that info from ETS - here also in a text form https://github.com/adorobis/xknx/blob/main/KNX%20AC%20group%20objects.txt).
There are separate group objects for controlling and reading status of the device.
Currently the Climate device works very well (some improvements in 0.116 HA version) allowing to control the A/C modes and setpoint temperature. I have also created a template sensor that is exposed to the FAN speed group address and nicely controls the FAN speed of the device. But it is not as nice as native integration and has to be done for each device.
```
 expose:
    - type: '1byte_unsigned'
      entity_id: 'sensor.ac_fan_speed_set_sypialnia'
      address: '1/4/9'
```
Let me know what else would be needed.

Cerize110, 2020-10-23:
Hello,
That would be a great improvement. I am in the same situation but with a Daikin A/C and a Zennio klic-dd-1.5 for the knx control
I have 5  informations:
1. on / off and state ->switch DPT 1.001 (ok in HA Climate)
2. target temperature and state-> temperature DPT 9.001 (ok in HA Climate)
3. modes -> ok in HA Climate
4. fan speed ans state -> percentage DPT 5.001 (0% ... 100% but 0% means AUTO)
5. swing and state -> switch DPT 1.001

I was not able to manage the fans speed currently in ha like Adorobis.
I don't know if this information helps ?
I am ready to help.

jncanches, 2020-11-18:
+1 for this request.

By the way the release of v15 included in HA 0.118 solved a lot of problem with KNX climate (my workaround was to use an MQTT climate and a very big flow in NodeRed to emulate the right behavior ~~)

So a very big thank you for this release !!!

nibblerrick, 2021-02-03:
Hi,

I am in the same boat. I made a messy node-red-flow to use the mqtt-hvac for translating to the knx bus because the climate-thing wasn't doing that much.
I have also Intesis-boxes but fot panasonic. The definition to the KNX-Bux seems to be the same.
I now looked again at the knx climate and yes, it works now with switching the climate.
But I also miss fan control and swing-modes.
@adorobis I also don't get how you use the exposed sensor for the fan, can you tell more about that?

farmio, 2021-02-03:
#582 lays the groundwork for a Fan Entity - afaik @crazyfx1 already started on an implementation.

crazyfx1, 2021-02-03:
Yes the fan entity will be added as soon as a new xknx version is published. The fan entity supports setting the speed in the first version.

adorobis, 2021-02-08:
Nice to see some new features, I'm afraid the "oscillation" function is not exactly what is needed for "swing mode" for A/C units. The oscillation can be `on` or `off` only whereas swing mode requires a few distinct values (in case of my Mitsubishi unit it has 5 positions and a swing/auto mode). So, the oscillation function would only allow to turn the auto mode on/off, not sure which position is selected when this is set to off. Would it be possible to add the vane position as well apart from the auto mode on/off? Or maybe it is already in the plans?

adorobis, 2021-02-08:
> @adorobis I also don't get how you use the exposed sensor for the fan, can you tell more about that?

You can add an `expose` section in your KNX configuration, below example for my two sensors which I use to control the A/C unit fan speed and room ambient temperature (for some reason my unit does not work well with that):
```
expose:
  - type: '1byte_unsigned'
    entity_id: 'sensor.ac_fan_speed_set_sypialnia'
    address: '1/4/9'
  - type: 'temperature'
    entity_id: 'sensor.aeotec_limited_zwa021_thermostatic_valve_temperature'
    address: '1/4/29'
```

crazyfx1, 2021-02-08:
The oscillation was mainly implemented like that, because HA also only supports oscillation on/off values. 
https://developers.home-assistant.io/docs/core/entity/fan/#oscillate

What KNX datatype is your oscillation and how do you set it?

adorobis, 2021-02-08:
> What KNX datatype is your oscillation and how do you set it?

Here is a definition for status and control telegrams:
```
Number	Name	Object Function	Description	Group Address	Length	C	R	W	T	U	Data Type	Priority
25	Control_ Vanes U-D Swing [DPT_1.002]	0-Off;1-Swing			1 bit	C	-	W	T	-	boolean	Low
71	Status_ Vanes U-D Swing [DPT_1.002]	0-Off;1-Swing			1 bit	C	R	-	T	-	boolean	Low
```

crazyfx1, 2021-02-08:
Thats just one boolean value? You said there are 5 values? Or are those other group addresses.

adorobis, 2021-02-08:
That is just for the on/off of the swing mode. The position can be set in a few ways, here is the full list of telegram definitions. The easiest way would be to use the `17	Control_ Vanes U-D / 5 Pos [DPT_5.010]	Position values: 1,2,3,4,5` for control and `63	Status_ Vanes U-D / 5 pos [DPT_5.010]	Position values: 1,2,3,4,5` for status.
```
Number	Name	Object Function	Description				Group Address	Length	C	R	W	T	U	Data Type	Priority
17	Control_ Vanes U-D / 5 Pos [DPT_5.010]	Position values: 1,2,3,4,5		1 byte	C	-	W	T	-	counter pulses (0..255)	Low
18	Control_ Vanes U-D / 5 Pos [DPT_5.001]	Thresholds:30%,50%,70% y 90%		1 byte	C	-	W	T	-	percentage (0..100%)	Low
19	Control_ Vanes U-D Man/Auto [DPT_1.002]	0-Manual;1-Auto				1 bit	C	-	W	T	-	boolean	Low
20	Control_ Vanes U-D Pos 1 [DPT_1.002]	1-Set Position 1			1 bit	C	-	W	T	-	boolean	Low
21	Control_ Vanes U-D Pos 2 [DPT_1.002]	1-Set Position 2			1 bit	C	-	W	T	-	boolean	Low
22	Control_ Vanes U-D Pos 3 [DPT_1.002]	1-Set Position 3			1 bit	C	-	W	T	-	boolean	Low
23	Control_ Vanes U-D Pos 4 [DPT_1.002]	1-Set Position 4			1 bit	C	-	W	T	-	boolean	Low
24	Control_ Vanes U-D Pos 5 [DPT_1.002]	1-Set Position 5			1 bit	C	-	W	T	-	boolean	Low
25	Control_ Vanes U-D Swing [DPT_1.002]	0-Off;1-Swing				1 bit	C	-	W	T	-	boolean	Low
63	Status_ Vanes U-D / 5 pos [DPT_5.010]	Position values: 1,2,3,4,5		1 byte	C	R	-	T	-	counter pulses (0..255)	Low
64	Status_ Vanes U-D / 5 pos [DPT_5.001]	20%, 40, 60%, 80% and 100%		1 byte	C	R	-	T	-	percentage (0..100%)	Low
65	Status_ Vanes U-D Man/Auto [DPT_1.002]	0-Manual;1-Auto				1 bit	C	R	-	T	-	boolean	Low
66	Status_ Vanes U-D Pos 1 [DPT_1.002]	1-Vanes in Position 1			1 bit	C	R	-	T	-	boolean	Low
67	Status_ Vanes U-D Pos 2 [DPT_1.002]	1-Vanes in Position 2			1 bit	C	R	-	T	-	boolean	Low
68	Status_ Vanes U-D Pos 3 [DPT_1.002]	1-Vanes in Position 3			1 bit	C	R	-	T	-	boolean	Low
69	Status_ Vanes U-D Pos 4 [DPT_1.002]	1-Vanes in Position 4			1 bit	C	R	-	T	-	boolean	Low
70	Status_ Vanes U-D Pos 5 [DPT_1.002]	1-Vanes in Position 5			1 bit	C	R	-	T	-	boolean	Low
71	Status_ Vanes U-D Swing [DPT_1.002]	0-Off;1-Swing				1 bit	C	R	-	T	-	boolean	Low
```

crazyfx1, 2021-02-08:
Well technically that would work with the new fan integration.
You will need to create a second knx fan entity where you use the group addresses connected to number 17 and number 63 for the state.
The first fan controls the speed.
The second fan controls the vanes position + swing mode as oscillation. Just use the speed attributes with the group addresses from the vanes position. The knx fan entity will be able to control DPT 5.010 by converting the UI fan position to steps. You will have to set the max_step attribute to 5.

Unfortunately it will be another month until the knx fan entity is available in HA.

adorobis, 2021-02-08:
OK, thanks for suggestion. I hope one day it will be implemented in the climate HA integration as it supports both [swing mode](https://www.home-assistant.io/integrations/climate/#service-climateset_swing_mode) and [fan modes](https://www.home-assistant.io/integrations/climate/#service-climateset_fan_mode) via respective services.

farmio, 2021-02-08:
This configuration seems to be only relevant for your specific device. 
I'd use a template and expose its state or create a script/automation with knx.send.

adorobis, 2021-02-08:
Well, not really. It uses an Intesis interface which is used by multiple A/C manufacturers for KNX integration. Additionally, as the HA Climate device allows for swing mode and fan mode control why not allowing it for the KNX climate? Anyway, a configuration would be needed to provide specific group addresses for control and status reading. So the question is which is most likely going to be the most generic approach, similarly to also quite generic configuration of fan levels in the fan entity.

nibblerrick, 2021-02-10:
> You can add an `expose` section in your KNX configuration, below example for my two sensors which I use to control the A/C unit fan speed and room ambient temperature (for some reason my unit does not work well with that):

Thanks @adorobis, I haven't had the idea I can send things to an exposed sensor, now it makes sense.

I would also like to see the fan and swing-modes directly in the climate-KNX-integration, as it is possible with HA as the mqtt-climate-entity shows it is very convenient. For the the datapoints are the same as the ones for @adorobis - I have panasonic-aplliances but they use the intesis-boxes for KNX, too.

marvin-w, 2021-04-12:
I agree with the OPs recommendation that for A/C units special circumstances apply that do not apply for "normal" floor heating.

Unfortunetly, this is really hard to implement when not having proper testing devices at home. Would any of you be able to test this in the custom integration if I find some spare time to implement this?

adorobis, 2021-04-12:
I would be more than happy to test the work in progress :)
I'm only interested to control the AC units independently from other heating devices I have (in my case it is central water heating with radiators in each room).
I'm mostly using heating and cooling modes, but devices support also Fan only, Dry and Auto modes - they are already workin ok in the current KNX integration. The only missing bit is Fan Speed and auto mode and Vanes position, swing and auto modes.

farmio, 2021-05-13:
There is now a PR for a climate template entity. Maybe this can be used to combine a KNX climate and fan entity easily.

shvirik, 2021-05-13:
> 
> 
> There is now a PR for a climate template entity. Maybe this can be used to combine a KNX climate and fan entity easily.
> [home-assistant/core#50107](https://github.com/home-assistant/core/pull/50107)

It's not perfect, but better than now. The problem with the fan entity, that it has "fan" limitations, different from the air conditioner one. Anyway, thanks a lot! :) I'll test it!

farmio, 2021-05-13:
What are these limitations? How would it need to behave differently?

shvirik, 2021-05-14:
> 
> 
> What are these limitations? How would it need to behave differently?

In advance, Beg my pardon for my terrible english.

As I said, I did the following correspondence:

off: Speed fan 1
low: Speed fan 2
med: Speed fan 3
high: Speed fan 4

And each position works, but I have a "strange" operation when pressing the buttons. I try to explain:

If the selected speed is 1 and I press 2 goes to 3. In this moment if i press 2 again goes 2.
If the selected speed is 1 and I press 3 goes to 4. In this moment if i press 3 again goes 3.
If the selected speed is 1 and I press 4 goes to 4. WTF! XD

If the selected speed is 2 and I press 1 goes to 1.
If the selected speed is 2 and I press 3 goes to 4. In this moment if i press 3 again goes to 3.
If the selected speed is 2 and I press 4 goes to 4.

If the selected speed is 3 and I press 1 goes to1. 
If the selected speed is 3 and I press 2 remains in 3. In this moment if i press 2 again goes to 2.
If the selected speed is 3 and I press 4 goes to 4

If the selected speed is 4 and I press 1 goes to 1.
If the selected speed is 4 and I press 2 goes to 3. In this moment if i press 2 again goes 2.
If the selected speed is 4 and I press 3 remains in 4. In this moment if i press 3 again goes to 3

It's disturbing, the first 8-10 times, but latter you know that some time you must push twice to arrive to the desirable speed.

Also, with the fan card you can choose between a 2 or 3 fan speed option (I have 4 because the off option works with the minimal air conditioner speed), but I don't know if the configuration could be set with a different value, and I think automatic mode is not available neither.

farmio, 2021-05-14:
@shvirik Im sure this can be solved if you could additionally provide a ETS groupmonitor log or xknx logs so we can see which payload is sent in your described cases and also which Status payload is replied then.
Maybe open a separate issue for this.

shvirik, 2021-05-14:
Thanks @farmio 

I've used the ETS %fan group address (and his state both 1byte address) with the home assistant knx address and state address:

[image]

[image]

I've checked the HomeAssistant knx fan address with DPT5.001 (percent value with 1byte) value mach with the ETS variable linked (% with 1byte too).

I tried monitorice the KNX BUS meanwhile I changed the fan speed, buy my knx ip interface only admit one connection concurrently , so ETS or HA could be used at the same time.

_"The interface cannot accept the new data connection because its maximum amount of concurrent connections is already used."_

I'll try to check the xknx events this weekend.

Regards,

farmio, 2021-05-14:
Why are there even 4 speeds? Wasn't this changed to a % slider recently for every fan entity? I guess the mapping of % - selected Speed is just off by a little.

shvirik, 2021-05-14:
> 
> 
> Why are there even 4 speeds? Wasn't this changed to a % slider recently for every fan entity? I guess the mapping of % - selected Speed is just off by a little.

Yes, if you click in the card fan icon, a the following card is opened with the speed slider and the oscilation switch.

[image]

If you move the button in the slide, the are the following correspondence:

0% -> Off, which I set as fan 1 which fortunately runs at the minimum speed of the air conditioner
33% -> low -> fan level 2
67% -> mid ->  fan level 3
100% -> high -> fan level 4

I've checked my aircondicioner controller  (https://www.zennio.com/products/clima/klic-mitt-v2), and you said is true. The maping speed is just off by a little... A litle big :)

[image]

But the values works for the correct levels,... except the first. 

1-25% -> level 1 (and the HA card give 0%) ???

adorobis, 2021-11-18:
unstale

nibblerrick, 2022-06-29:
As it's warm again and the climate-devices are more often used I wanted to kindly ask if there could be some progress in the near future or not?

adorobis, 2022-11-23:
Hi, is there any hope to have swing mode and fan speed/mode implemented for KNX Climate?

farmio, 2022-11-23:
My xknx todo list is quite full, I personally don't plan to work on any new features in device modules in foreseeable future. 
I'm happy to review code though.

adorobis, 2022-11-23:
Thanks for info! I'll have a look at how complex it would be to add it but I'm afraid my skills might not be good enough :)

DirkMu, 2023-05-15:
Hello ✋ is there any chance that this can be implemented. 

I am using the current climate and fan implementation which does not cover the use case. The fan supports on/off respectively percentage and oscillation. Typicalle in combination with an AC the fan supports setting the position of the vanes (up and down vanes) and using mode auto. In addition, when the AC is turned off the fan remains on. This is because the AC actuator will not send an off or 0% as a response to turning the unit off or setting the HVAC mode to off.  Thus, the fan remains on. I am aware that I can set up an automation in HA that turns the fan off when the AC goes off 😉

I would like to control my AC unit right from the climate entity since the climate can controll fans, too. This can be seen in the MelCloud integration (Mitsubishi AC). Since the integration requires internet access it is very slow and sometimes not reachable. 

I am not able to do the development. I can provided information as much as possible if required. Please let me know if we can support.

sanketgroup, 2023-08-21:
I hope, there will be options to integrate Fan speed and Van position properly in XKNX HA.

I see this request post is almost 3 years old.
It would be really helpful for community.

Team XKNX (@farmio , @marvin-w , @crazyfx1) Rocks

wagnst, 2023-11-04:
+1 to add this into the `climate` function. All modern AC units meanwhile have up/down and left/right swing configurable (fixed positions as well as auto) as well as fan speeds (steps & auto).

zhujy09, 2023-11-07:
+1 to add this into the climate function. 
Fan Speed and Swing mode control are quite basic functions in all the A/C devices.

thomaszlow, 2023-11-17:
+1 also from my side for fan speed, horizontal and vertical oscillation

arnocl, 2024-01-04:
+1 for this feature request. The fan is an integral part of the HVAC device, it should also be part of the climate entity instead of being a different fan device.

mechanliang, 2024-06-19:
附议要求在KNX气候集成中添加风速档位调节

svarco, 2024-07-19:
+1
I have a Daikin "Stylish" AC with a Zennio KLIC-DD v3 interface! 🙂

closed 2024-09-18 by marvin-w

wagnst, 2024-09-30:
@marvin-w sorry but i believe the referenced PR does not actually solve (all) the discussed features in this issue.
It only adds one functions to control fan speed basically.  

My Air Conditions not only have fan speed, but also directions of valves (left/right, up/down) which can be set to a static position, auto or off. See also https://github.com/XKNX/xknx/issues/466#issuecomment-1793419336.

Kindly re-open this issue again as from my perspective only the fan speed functionality would be supported by this, but not the rest of the functionality that most modern AC have and are respectively exposing (e.g. see the Gree HA integration which supports all those mentioned features also. See for example here: https://github.com/RobHofmann/HomeAssistant-GreeClimateComponent/blob/master/custom_components/gree/climate.py#L83C290-L84C1

PR #1572 (2024-09-18): add support for fan_speed on climate device

## Description

Add fan speed support to climate device. 

Fixes #466

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [X] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Code quality improvements to existing code or addition of tests

## Checklist

- [X] The documentation has been adjusted accordingly
- [X] Tests have been added that prove the fix is effective or that the feature works
- [X] The chang [... cut]
PR #1642 (2025-01-25): Add swing and horizontal swing to climate device

## Description
Add swing and horizontal swing support to climate device

Fixes #466

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Code quality improvements to existing code or addition of tests

## Checklist

- [x] The documentation has been adjusted accordingly
- [x] Tests have been added that prove the fix is effective or that the feature wo [... cut]