XKNX/xknx #481: cover.set_cover_position not stopping with no position_address

Janhsen, 2020-10-25:
**Description of problem:**

Dear xknx team,

covers with **no position_address** do not longer stop when moving the slider in the ha frontpanel to a position e.g. 30%. I attached my according HA config as well as some debug log from HA. This was working in precious versions of Home Assistant for me ( version 0.111).

The **service cover.set_cover_position ist not stopping at this value**. The services cover.close_cover;  service cover.open_cover;  cover.stop_cover are working currently.

Maybe a bug to the reworked travelcalculator / cover.py? It seems to me, that there is no stop movement send to actor when the calculated position is reached.

Remark: My covers do not have a positional feedback. But this applies not only to my knx config, this reproducible to other knx hardware setup with HA as well (a friend of mine reproduces this bug with his knx covers (which have a position_address). When deleting the position address in his config, this bug occurs as well. It would be great, to have this feature back / working again! Let me know, if i can assist you reproducing this bug. 

All the best 
Jan

- [x ] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:  0.116.4
- last working xknx / Home-Assistant release (if known): 0.111

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
[LOG_ha.txt](https://github.com/XKNX/xknx/files/5434327/LOG_ha.txt)
[knx_cover.yaml.txt](https://github.com/XKNX/xknx/files/5434337/knx_cover.yaml.txt)
[knx.yaml.txt](https://github.com/XKNX/xknx/files/5434338/knx.yaml.txt)
[image]

closed 2020-10-26 by farmio

BjHampel, 2020-11-05:
Hi
I have the same problem with my KNX covers. 
In an older version of hass.io (don't know which exactly) the "cover.set_cover_position" moved the cover and stopped automaticaly at defined position.
After updating to the newest version, the cover is starting to move in the correct direction, but isn't stopping anymore so that it runs to 0% or 100%.

the commands "cover.open_cover", "cover.close_cover" and "cover.stop_cover" are working fine.

Do I have to change anything in my KNX system, or is this problem related to HomeAssistant?
and: any recommendations howto fix this problem so that it works again?

BR
Björn

the cover is defined as:
knx:
  tunneling:
    host:  !secret knx_host
    port:  !secret knx_port
    local_ip:  !secret  knx_local_ip
  cover: 
    - name: OG WC_Rolladen Fenster
      move_long_address: '2/2/41'
      move_short_address: '2/2/50'
      travelling_time_down: 15
      travelling_time_up: 17

farmio, 2020-11-05:
Hi!
You'll have to wait until xknx 0.15.3 is shipped with HA.
See the linked PR above.

PR #482 (2020-10-26): Cover: Fix auto-stopping without position state

## Description

Fixes auto-stopping covers without position state. Due to processing every (outgoing) telegram in every device we did overwrite the target position with the end position. 

Eg. when we want from closed to 50% an UP-Telegram is sent to the bus (because we have no position address) and when reaching 50% a stop() shall be called. When processing this (outgoing) UP-Telegram we were treating it like any UP so target position became 0%.

Now we check if the cover is already moving in the received direction and ignore the telegram if it [... cut]