XKNX/xknx #486: Sensors won't work

sagitt, 2020-10-28:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: n/a
- last working xknx / Home-Assistant release (if known): n/a

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

After upgrade to home assistant 0.117 sensor before working, now won't work.. like energy meter

[image]

```
    # Energia [Rilevato]
    - name: 'Energia [Potenza Attiva]'
      state_address: '7/0/0'
      sync_state: true
      type: 'power'
    - name: 'Energia [Potenza Reattiva]'
      state_address: '7/0/1'
      sync_state: true
      type: 'electric_potential'
    - name: 'Energia [Tensione]'
      state_address: '7/0/2'
      sync_state: true
      type: 'electric_potential'
    - name: 'Energia [Corrente]'
      state_address: '7/0/3'
      sync_state: true
      type: 'electric_current'
    - name: 'Energia [Fattore Di Potenza]'
      state_address: '7/0/4'
      sync_state: true
      type: 'powerfactor'
    - name: 'Energia [Frequenza]'
      state_address: '7/0/5'
      sync_state: true
      type: 'frequency'
```

farmio, 2020-10-29:
🔮 Please provide more information. 
- Which version were you running before, when it worked?
- Paste log messages! Setting xknx.log to debug level would probably help. 
- Does anything else work? Eg sending to bus, updating states of a light etc.

sagitt, 2020-10-29:
> 🔮 Please provide more information.
> 
> * Which version were you running before, when it worked?
> * Paste log messages! Setting xknx.log to debug level would probably help.
> * Does anything else work? Eg sending to bus, updating states of a light etc.

Last really working Hass is 0.114,

0.115 add binary sensor bug, disconnections bug, bad warning bug (https://github.com/XKNX/xknx/issues/474)
0.116 fix binary sensor bug, disconnections and bad warning still
0.117 disconnections to test, bad warning still and no energy meter sensor works

yes, other "sensors" like my covers %, "binary sensors" like alarm works. only energy meter won't work. downgrading to 0.116 make it works. for example image below, are "sensors". Turn on\off lights, up\down\% covers etc. works fine. i not tested every configured item but these works.

```
sensor:
    # Tapparelle
    # Cucina
    - name: 'Stato Tapparella [Cucina]'
      state_address: '2/2/0'
      sync_state: 'init'
      type: 'percent'
    # Soggiorno
    - name: 'Stato Tapparella [Soggiorno]'
      state_address: '2/2/1'
      sync_state: 'init'
      type: 'percent'
    # Camera 3
    - name: 'Stato Tapparella [Camera 3]'
      state_address: '2/2/2'
      sync_state: 'init'
      type: 'percent'
    # Bagno
    - name: 'Stato Tapparella [Bagno]'
      state_address: '2/2/3'
      sync_state: 'init'
      type: 'percent'
    # Camera 2
    - name: 'Stato Tapparella [Camera 2]'
      state_address: '2/2/4'
      sync_state: 'init'
      type: 'percent'
    # Camera 1
    - name: 'Stato Tapparella [Camera 1]'
      state_address: '2/2/5'
      sync_state: 'init'
      type: 'percent'

```
[image]

i'll make a log soon.

sagitt, 2020-10-29:
[home-assistant.log](https://github.com/XKNX/xknx/files/5457296/home-assistant.log)

this is the log

farmio, 2020-10-29:
Your energy meters report values of 0. There is nothing we can do. Maybe have a look with ETS diagnostics if you get the same values there, but I couldn't see how this would differ.
```python
# 7/0/0
2020-10-29 10:18:25 DEBUG (MainThread) [xknx.raw_socket] Received from ('192.168.0.30', 3671): 061004200019040268002900bce01106380005004000000000
2020-10-29 10:18:25 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_REQUEST" Reserve="0" TotalLength="25" />
 body="<TunnellingRequest communication_channel_id="2" sequence_counter="104" cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.6")" DestinationAddress="GroupAddress("7/0/0")" Flags="1011110011100000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTArray value="[0x0,0x0,0x0,0x0]" />" />" />" />

# 7/0/1
2020-10-29 10:18:25 DEBUG (MainThread) [xknx.raw_socket] Received from ('192.168.0.30', 3671): 061004200015040267002e00bce011663801010000
2020-10-29 10:18:25 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_REQUEST" Reserve="0" TotalLength="21" />
 body="<TunnellingRequest communication_channel_id="2" sequence_counter="103" cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.102")" DestinationAddress="GroupAddress("7/0/1")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="<DPTBinary value="0" />" />" />" />
```
They don't seem to send on their own, only respond to the state updater - at least in the 2 minutes of logs you provided.

sagitt, 2020-10-29:
> Your energy meters report values of 0. There is nothing we can do. Maybe have a look with ETS diagnostics if you get the same values there, but I couldn't see how this would differ.
>
> ```python
> # [...]

ok if the problem is in my hardware i'll try to fix it.

closed 2020-11-02 by sagitt