XKNX/xknx #491: Feature: Optimistic Mode / State assumption

KNXBroker, 2020-11-10:
Dear all,

for the time being xKNX always uses OPTIMISTIC MODE, which means that the status of lights, switches and covers updates always automatically, it does not matter whether there is any state object or not.

However the behaviour of lovelace is different if an entity is in optimstic mode or not. If optimistic mode is on:
-  no switch is displayed any more for switches and lights, however two buttons for switching on and off.
-  the cover arrows for UP and DOWN are shown always and are not getting hidden (the feature I was missing). 

Therefore I would propose to implement Optimistic Mode / State assumption in xKNX as well.  What do you think?

For switches and lights the change should be pretty easy. If there is no state object defined, optimistic mode will be used automatically. In addition the user shall be able to manually overriede optimistic mode. Is this a breaking change?  Home assistant user would get their visualisation changed.

For covers I'm not quite sure how to implement the feature. The easiest would be to do it identically to switches and lights. However, for the time being, xKNX is updating the position of a cover always with the time parameters. To be persistent, switching to `optimistic: false` would mean disabling this feature as well, which makes no sense in my opinion. What do you think?

And last but not least, I'm not sure about the wording. MQTT and Template are using `optimistic`, however the attribute is called `assumed_state`. What do you think?

Looking forward to hearing from you.

farmio, 2020-11-14:
I like this for switches. Could be useful for trigger buttons. Is there a way to disable one or the other button too - to have just a "On" or "Off" button like a binary scene? This would be even cooler.
[image]
We could also start with `Device.state = None` instead of `False` and use 
```python
    @property
    def assumed_state(self) -> bool:
        """Return True if unable to access real state of the entity."""
        return self._device.state is None
```
This would be nice for covers to help set the inital direction when we don't have a position_state_address.

With lights I'm not sure about the usefulness. One could just use a switch if it was really needed, with lights we should know the state, brightness, color, etc.

KNXBroker, 2020-11-17:
@farmio , we just had an discussion regarding native icon support #495. Isn't this topic more or less the same? `assumed_state ` can defined by customizing as well, but - for me - it is again not end user friendly ;-) Looking forward for your comments.

farmio, 2020-11-17:
Thanks, I didn't know that. So we could leave this PR as is (assumed_state only when state is not yet known implemented only in the HA integration code) and don't have to add config options 😃

farmio, 2020-11-22:
sadly this isn't allowed in HA https://github.com/home-assistant/core/pull/43536#discussion_r528406212

farmio, 2021-01-07:
Cover initialization state in HA should be fixed now. 
For switches the HA customize key should be sufficient.

closed 2021-01-07 by farmio