XKNX/xknx #497: Enhance HA knx_event

farmio, 2020-11-17:
There are some requests about updating the HA knx_event. 
Eg. https://github.com/home-assistant/core/issues/43256

Currently we just fire the `knx_event` for incoming telegrams, as stated in the documentation.

We could
- fire on outgoing telegrams too, not only on incoming
- add a parameter "direction" for incoming/outgoing
- add a parameter "source" for IA (eg. to identify which wall switch was used)

I'm still not very convinced about using the `knx_event` at all since we can't really decode the payload and dealing with raw payloads should not be a user-facing task in HA (besides maybe for DPT 1/2/3).

cassianyoung, 2020-12-15:
Hi, did this go into HA 2012.12 ?

farmio, 2020-12-15:
No

PR #572 (2021-02-16): Feat: Also run callbacks on outgoing telegrams, but filter them out b…

## Description

Do trigger telegram callbacks also for outgoing telegrams for HA. For all other callbacks this is disabled by default and has to be explicitly enabled.

Fixes #497 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

## Checklist:

- [x] I have performed a self-review [... cut]
closed 2021-02-16 by farmio