XKNX/xknx #499: KNX bus did not respond in time (1 secs) to GroupValueRead request for: x/x/x errors

FredericMa, 2020-11-19:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

On HA version 0.118.0 I see a lot of `KNX bus did not respond in time (1 secs) to GroupValueRead request for: x/x/x errors`.
On 0.117 I only saw a few at startup but now they also appear every hour during the refresh.

**Version information:**
- xknx / Home-Assistant release with the issue: 0.118.0
- last working xknx / Home-Assistant release (if known): 0.117.1 (had less warnings in the logs)

**KNX installation:**

MDT IP Router

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```yaml
knx: 
  routing:
    local_ip: xxx.xxx.xxx.xxx

  light:
    ## Ground floor

    # Lights
    - name: storage # LTL
      address: '0/0/10'
      state_address: '3/1/9'
```
**Traceback (if applicable):**
Filtered log with one example (from startup):
```
2020-11-19 11:01:39 DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueSwitch device_name="storage" feature_name="State" GroupAddress("0/0/10")/GroupAddress("3/1/9")/None/None/>
2020-11-19 11:01:40 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/9 for storage - State
2020-11-19 11:01:41 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("3/1/9")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
2020-11-19 11:01:41 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.ROUTING_INDICATION" Reserve="0" TotalLength="17" />
 body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("15.15.250")" DestinationAddress="GroupAddress("3/1/9")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
2020-11-19 11:01:41 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time (1 secs) to GroupValueRead request for: 3/1/9
2020-11-19 11:01:41 WARNING (MainThread) [xknx.log] Could not sync group address '3/1/9' (storage - State)
2020-11-19 11:01:41 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.ROUTING_INDICATION" Reserve="0" TotalLength="17" />
 body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.5")" DestinationAddress="GroupAddress("3/1/9")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTBinary value="0" />" />" />" />
2020-11-19 11:01:42 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("3/1/9")", payload="<DPTBinary value="0" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
```

I'll update later to add a debug log after the hour refresh.

DirkMu, 2020-11-19:
Same issue in my installation. Around 120 log entries per hour (with every state updater schedule). However, I do not face the issue that GA state addresses are read twice.

farmio, 2020-11-19:
@FredericMa can you please add the configuration of the devices that are read twice?

FredericMa, 2020-11-19:
Sure! Don't mind the `# LTL` comments, I use it to mark my long term logged entities.

Light (from example but others are similar):
```yaml
    - name: garage # LTL
      address: '0/0/24'
      state_address: '3/1/15'
```

Thermostats:
```yaml
  climate:
    - name: bedroom_front # LTL
      temperature_address: '4/0/0'
      target_temperature_address: '1/0/0'
      target_temperature_state_address: '4/1/0'
      operation_mode_address: '1/1/0'
      operation_mode_state_address: '4/2/0'
      min_temp: 7.0
      max_temp: 28.0
      operation_modes: ['Comfort', 'Frost Protection']
      controller_modes: ['Heat']

    - name: office # LTL
      temperature_address: '4/0/1'
      target_temperature_address: '1/0/1'
      target_temperature_state_address: '4/1/1'
      operation_mode_address: '1/1/1'
      operation_mode_state_address: '4/2/1'
      min_temp: 7.0
      max_temp: 28.0
      operation_modes: ['Comfort', 'Frost Protection']
      controller_modes: ['Heat']

    - name: bathroom # LTL
      temperature_address: '4/0/2'
      target_temperature_address: '1/0/2'
      target_temperature_state_address: '4/1/2'
      operation_mode_address: '1/1/2'
      operation_mode_state_address: '4/2/2'
      min_temp: 7.0
      max_temp: 28.0
      operation_modes: ['Comfort', 'Frost Protection']
      controller_modes: ['Heat']

    - name: bedroom_side # LTL
      temperature_address: '4/0/3'
      target_temperature_address: '1/0/3'
      target_temperature_state_address: '4/1/3'
      operation_mode_address: '1/1/3'
      operation_mode_state_address: '4/2/3'
      min_temp: 7.0
      max_temp: 28.0
      operation_modes: ['Comfort', 'Frost Protection']
      controller_modes: ['Heat']

    - name: master_bedroom # LTL
      temperature_address: '4/0/4'
      target_temperature_address: '1/0/4'
      target_temperature_state_address: '4/1/4'
      operation_mode_address: '1/1/4'
      operation_mode_state_address: '4/2/4'
      min_temp: 7.0
      max_temp: 28.0
      operation_modes: ['Comfort', 'Frost Protection']
      controller_modes: ['Heat']
```

FredericMa, 2020-11-19:
@DirkMu did you also enable debug logging for xknx? The duplicate messages appear as debug loggings and its only at the refresh, so not at startup.

FredericMa, 2020-11-19:
Another thing I noticed is that my switches are out of sync after a restart while lights are correctly synced at startup.

The only difference is the domain:
```yaml
  light:
    - name: front_door # LTL
      address: '0/0/29'
      state_address: '3/1/25'
  switch:
    - name: christmas_tree # LTL
      address: '0/3/0'
      state_address: '3/3/0'
```

Logs:
Christmas tree `switch` appears as off while it is on (oh yes, christmas tree is already in place):
```
	Line 62: 2020-11-19 20:32:09 DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueSwitch device_name="christmas_tree" feature_name="State" GroupAddress("0/3/0")/GroupAddress("3/3/0")/None/None/>
	Line 623: 2020-11-19 20:32:17 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/3/0 for christmas_tree - State
	Line 624: 2020-11-19 20:32:17 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("3/3/0")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
	Line 626:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("15.15.250")" DestinationAddress="GroupAddress("3/3/0")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
	Line 645:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.4")" DestinationAddress="GroupAddress("3/3/0")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTBinary value="1" />" />" />" />
	Line 646: 2020-11-19 20:32:17 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("3/3/0")", payload="<DPTBinary value="1" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
```
Front door `light` appears as on while it is on. == correct.
```
	Line 46: 2020-11-19 20:32:09 DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueSwitch device_name="front_door" feature_name="State" GroupAddress("0/0/29")/GroupAddress("3/1/25")/None/None/>
	Line 495: 2020-11-19 20:32:16 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/25 for front_door - State
	Line 500: 2020-11-19 20:32:16 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("3/1/25")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
	Line 502:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("15.15.250")" DestinationAddress="GroupAddress("3/1/25")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
	Line 514:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.5")" DestinationAddress="GroupAddress("3/1/25")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTBinary value="1" />" />" />" />
	Line 515: 2020-11-19 20:32:16 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("3/1/25")", payload="<DPTBinary value="1" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
```

Should I open another issue for this one?

DirkMu, 2020-11-19:
@FredericMa no. log level is warning. 

I also noticed states to be out of sync. Yesterday, right after the installation of 0.118 cover states were out of sync. Since then I did not restart HA. A few minutes ago I realized the state of a light was not in sync. It was switched on and while it was switched off the state_updater ran in parallel and the mentioned errors occurred. The off state was not received in HA. Thus the light in HA was still on while it was off in fact.

FredericMa, 2020-11-23:
I upgraded my HA to use version 0.15.4 and here are my findings:

- Switch states work OK again.
- I still see a lot of duplicate state reads every hour:
```
	Line 4455: 2020-11-23 09:54:34 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/9 for storage - State
	Line 4456: 2020-11-23 09:54:34 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/9 for storage - State
	Line 4471: 2020-11-23 09:54:35 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/4 for kitchen_stove - State
	Line 4472: 2020-11-23 09:54:35 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/4 for kitchen_stove - State
	Line 4487: 2020-11-23 09:54:35 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/5 for corridor_gf - State
	Line 4495: 2020-11-23 09:54:35 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/12 for toilet_gf - State
	Line 4503: 2020-11-23 09:54:36 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/15 for garage - State
	Line 4504: 2020-11-23 09:54:36 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/15 for garage - State
	Line 4519: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/0 for living_room_sofa - State
	Line 4520: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/0 for living_room_sofa - State
	Line 4536: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/0 for living_room_sofa - Brightness
	Line 4537: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/0 for living_room_sofa - Brightness
	Line 4552: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/1 for living_room_dresser - State
	Line 4553: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/1 for living_room_dresser - State
	Line 4568: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/1 for living_room_dresser - Brightness
	Line 4569: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/1 for living_room_dresser - Brightness
	Line 4580: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/10 for living_room_tv_table - State
	Line 4581: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/10 for living_room_tv_table - State
	Line 4600: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/3 for living_room_tv_table - Brightness
	Line 4601: 2020-11-23 09:54:37 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/3 for living_room_tv_table - Brightness
	Line 4612: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/16 for living_room_dining_table - State
	Line 4620: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/6 for living_room_dining_table - Brightness
	Line 4628: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/20 for living_room_wall - State
	Line 4636: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/20 for living_room_wall - State
	Line 4644: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/7 for living_room_wall - Brightness
	Line 4645: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/7 for living_room_wall - Brightness
	Line 4660: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/11 for kitchen_island - State
	Line 4661: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/1/11 for kitchen_island - State
	Line 4676: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/4 for kitchen_island - Brightness
	Line 4677: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/2/4 for kitchen_island - Brightness
	Line 4692: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/6 for living_room_dresser_strip - State
	Line 4693: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/6 for living_room_dresser_strip - State
	Line 4708: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/7 for living_room_dresser_strip - Color RGB
	Line 4709: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/7 for living_room_dresser_strip - Color RGB
	Line 4724: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/3 for living_room_sofa_strip - State
	Line 4725: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/3 for living_room_sofa_strip - State
	Line 4740: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/4 for living_room_sofa_strip - Color RGB
	Line 4741: 2020-11-23 09:54:38 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/4 for living_room_sofa_strip - Color RGB
	Line 4756: 2020-11-23 09:54:39 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/0 for living_room_tv_strip - State
	Line 4757: 2020-11-23 09:54:39 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 3/4/0 for living_room_tv_strip - State
[... cut ...]
2020-11-23 09:54:43 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.ROUTING_INDICATION" Reserve="0" TotalLength="17" />
 body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.4")" DestinationAddress="GroupAddress("3/3/1")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTBinary value="0" />" />" />" />
2020-11-23 09:54:43 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("3/3/1")", payload="<DPTBinary value="0" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
```

PR #506 (2020-11-24): Fix StateUpdater spawning 2 update loops

## Description

always close the update task before starting a new

Fixes #499

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

## Checklist:

- [x] I have performed a self-review of my own code
- [ ] The documentation has been adjusted accordingly
- [x] The changes generate no new warnings
- [ ] Tests have be [... cut]
closed 2020-11-24 by farmio