XKNX/xknx #500: Climate operation mode set to Away after HA restart

FredericMa, 2020-11-19:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

The operation mode of all climate entities is set to away even though the states are correctly fetched from the devices at startup of HA.
This issue was intended to be fixed via #460 like mentioned in this comment https://github.com/home-assistant/core/issues/40768#issuecomment-717083946 but I still see this happening
It is reading `4` 'Frost Protection' from the bus but yet it is set to `Away`.

**Version information:**
- xknx / Home-Assistant release with the issue: 0.15.3/0.118.0
- last working xknx / Home-Assistant release (if known): 0.117.1 (had less warnings in the logs)

**KNX installation:**

MDT IP Router

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```yaml
knx: 
  routing:
    local_ip: xxx.xxx.xxx.xxx

  climate:
    - name: bathroom
      temperature_address: '4/0/2'
      target_temperature_address: '1/0/2'
      target_temperature_state_address: '4/1/2'
      operation_mode_address: '1/1/2'
      operation_mode_state_address: '4/2/2'
      min_temp: 7.0
      max_temp: 28.0
      operation_modes: ['Comfort', 'Frost Protection']
      controller_modes: ['Heat']
```
**Traceback (if applicable):**
Filtered log with one example (from startup):
```
2020-11-19 11:01:39 DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueClimateMode device_name="bathroom Mode" feature_name="Operation mode" GroupAddress("1/1/2")/GroupAddress("4/2/2")/None/None/>
2020-11-19 11:01:50 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 4/2/2 for bathroom Mode - Operation mode
2020-11-19 11:01:50 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/2")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
 body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("15.15.250")" DestinationAddress="GroupAddress("4/2/2")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
2020-11-19 11:01:50 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.ROUTING_INDICATION" Reserve="0" TotalLength="18" />
 body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.27")" DestinationAddress="GroupAddress("4/2/2")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTArray value="[0x4]" />" />" />" />
2020-11-19 11:01:50 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/2")", payload="<DPTArray value="[0x4]" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
```
[image]

farmio, 2020-11-19:
Does it change anything when you don't use custom modes list?

> ```yaml
>       # operation_modes: ['Comfort', 'Frost Protection']
>       # controller_modes: ['Heat']
> ```

FredericMa, 2020-11-19:
> Does it change anything when you don't use custom modes list?

No, it doesn't change it but I did notice something weird.
I did a restart after changing 2 thermostats with your suggested change (bedroom_front and office).
After the restart one came up correctly and the other one not. I checked the operation_mode_state_address logs and this came up:
```
away
Search "4/2/0" (6 hits in 1 file of 8 searched)
  D:\HomeAssistant\Config\home-assistant.log (6 hits)
	Line 47: 2020-11-19 20:21:53 DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueClimateMode device_name="bedroom_front Mode" feature_name="Operation mode" GroupAddress("1/1/0")/GroupAddress("4/2/0")/None/None/>
	Line 504: 2020-11-19 20:21:59 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 4/2/0 for bedroom_front Mode - Operation mode
	Line 505: 2020-11-19 20:21:59 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/0")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
	Line 507:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("15.15.250")" DestinationAddress="GroupAddress("4/2/0")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
	Line 526:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.25")" DestinationAddress="GroupAddress("4/2/0")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTArray value="[0x4]" />" />" />" />
	Line 527: 2020-11-19 20:21:59 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/0")", payload="<DPTArray value="[0x4]" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
	
eco
Search "4/2/1" (8 hits in 1 file of 8 searched)
  D:\HomeAssistant\Config\home-assistant.log (8 hits)
	Line 50: 2020-11-19 20:21:53 DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueClimateMode device_name="office Mode" feature_name="Operation mode" GroupAddress("1/1/1")/GroupAddress("4/2/1")/None/None/>
	Line 528: 2020-11-19 20:21:59 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 4/2/1 for office Mode - Operation mode
	Line 537: 2020-11-19 20:22:00 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/1")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
	Line 539:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("15.15.250")" DestinationAddress="GroupAddress("4/2/1")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
	Line 542: 2020-11-19 20:22:00 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time (1 secs) to GroupValueRead request for: 4/2/1
	Line 543: 2020-11-19 20:22:00 WARNING (MainThread) [xknx.log] Could not sync group address '4/2/1' (office Mode - Operation mode)
	Line 653:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.26")" DestinationAddress="GroupAddress("4/2/1")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTArray value="[0x4]" />" />" />" />
	Line 654: 2020-11-19 20:22:00 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/1")", payload="<DPTArray value="[0x4]" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
```
Note that bedroom_front doesn't show any error, receives the correct data but show up as `away` which is wrong.
Office shows an error (the same as discussed in #499), also receives the correct data and shows the correct state `eco`.

After that I did another restart, both showed no error, received correct data but show both the wrong state `away`:
```
away
Search "4/2/0" (6 hits in 1 file of 1 searched)
  D:\HomeAssistant\Config\home-assistant.log (6 hits)
	Line 47: 2020-11-19 20:26:18 DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueClimateMode device_name="bedroom_front Mode" feature_name="Operation mode" GroupAddress("1/1/0")/GroupAddress("4/2/0")/None/None/>
	Line 702: 2020-11-19 20:26:25 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 4/2/0 for bedroom_front Mode - Operation mode
	Line 711: 2020-11-19 20:26:25 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/0")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
	Line 713:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("15.15.250")" DestinationAddress="GroupAddress("4/2/0")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
	Line 719:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.25")" DestinationAddress="GroupAddress("4/2/0")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTArray value="[0x4]" />" />" />" />
	Line 720: 2020-11-19 20:26:25 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/0")", payload="<DPTArray value="[0x4]" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />

away
Search "4/2/1" (6 hits in 1 file of 1 searched)
  D:\HomeAssistant\Config\home-assistant.log (6 hits)
[... cut ...] />" />" />
	Line 750:  body="<RoutingIndication cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.26")" DestinationAddress="GroupAddress("4/2/1")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTArray value="[0x4]" />" />" />" />
	Line 751: 2020-11-19 20:26:25 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/2/1")", payload="<DPTArray value="[0x4]" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
```

FredericMa, 2020-11-25:
I can confirm that #503 fixes this issue.
Thanks!

closed 2020-11-25 by FredericMa

alphasixtyfive, 2024-08-19:
In my case it started switching to "Sleep" after restart. I'm not sure if it started happening after update to HA 2024.8 or before that. 

[image]

```
  climate:
    - name: Hallway
      temperature_step: 0.5
      temperature_address: 4/6/1
      target_temperature_state_address: 4/5/1
      setpoint_shift_address: 3/3/1
      setpoint_shift_state_address: 4/3/1
      operation_mode_address: 3/1/1
      controller_status_state_address: 4/0/1
      operation_modes:
        - Comfort
        - Standby
        - Night
        - Frost Protection
```

I've tried commenting out `operation_mode` but that didn't do anything apart from adding "None" as a mode.

farmio, 2024-08-19:
@alphasixtyfive I don't think this is related to the original, 4 year old, issue. Please open new issues for new problems next time. 

Your GA 4/0/4 you posted isn't used in the configuration of your example so it's hard to say what that is.

alphasixtyfive, 2024-08-19:
> @alphasixtyfive I don't think this is related to the original, 4 year old, issue. Please open new issues for new problems next time. 
> 
> Your GA 4/0/4 you posted isn't used in the configuration of your example so it's hard to say what that is. 

Thanks, I'll open a new ticket. I've changed it to 4/0/1 in my comment recently.