XKNX/xknx #507: knx: state changes of pir (binary) sensors not always processed by HA. Random behaviour

Holland1, 2020-11-24:
**Description of problem:**

- [ ] using xknx standalone
- [ x] using Home-Assistant knx plugin

Just wondered if you guys have already seen this issue: https://github.com/home-assistant/core/issues/43414, or is this issue already resolved in 0.15.4?
For some reason the issue was not picked up by the probot, that's the reason that I'm forwarding this now

**Version information:**
- xknx / Home-Assistant release with the issue: 0.118.2
- last working xknx / Home-Assistant release (if known): 0.115

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

**Traceback (if applicable):**

farmio, 2020-11-24:
Yes, we see this, don't worry.

marvin-w, 2020-11-28:
Duplicate.

closed 2020-11-28 by marvin-w