XKNX/xknx #521: Feature request : Cover remote values in Home Assistant & Possibility to restore their state in Home Assistant

CharlieBailly, 2020-12-05:
Dear all,
Unfortunately, there are still KNX covers with no state addresses, which means Home Assistant only knows the real state when a move_address is on the KNX bus. There are two problems here according to me :

1. Only one move_address can change this state and if you have multiple buttons that pulls up or down your cover with different move_addresses, Home Assistant will not update the state of the cover when using these buttons.
2. Even if you are lucky and you have only one move_address that your cover is listening to on the KNX bus, if you restart Home Assistant, the state of the cover is lost since it is not recovered by Home Assistant since the cover do not have a position_address.

I am actually using a custom xknx version in Home Assistant to solve theses two issues, do you think it is worth it to make a pull request to add theses functionalities ?

Here is a sample configuration I am using :

```yaml
xknx:
  ...
  ...
  cover:
    - name: "Volet petite chambre"
      travelling_time_up: !secret travelling_time_up_large
      travelling_time_down: !secret travelling_time_down_large
      move_long_address: "2/4/0" #This one is used to actually send the KNX payload to move up and down the cover
      stop_address: "2/4/1" #This one is used to actually send the KNX payload to stop the cover
      move_long_remote_addresses: #Home assistant is listening to these adresses to update the cover state (up/down)
        - "24/4/23"
        - "10/4/226"
        - "24/4/45"
      stop_remote_addresses: #Home assistant is listening to these adresses to update the cover state (stop)
        - "24/4/24"
        - "10/4/225"
        - "24/4/46"

```
I am also implicitly Restoring the state of the covers (they inherit from the RestoreEntity in Home Assistant in my custom xknx components) but maybe we could add a boolean value in the cover config as well such as :

```yaml
restore_value: true # or false (and maybe true by default if you use move_long_remote_addresses/stop_remote_addresses )
```
Tell me what do you think of it ?
Thanks

farmio, 2020-12-05:
This can be done with `passive_addresses` in xknx (not supported from HA yet). 

How does the value restoration work? Is there a timeout to not do it on longer downtimes of HA? If a cover was moved while HA was down it would have wrong states. 

I'd customize covers without any state with `assumed_state: True`. Unfortunately this isn't allowed to change on runtime (to disable it when we know state again).

CharlieBailly, 2020-12-05:
Yes, I am using `passive_addresses` in my custom xknx component, because you told me to try it [here](https://community.home-assistant.io/t/custom-component-cover-time-based/187654/84?u=charliebailly) and it worked, do you remember ? But I am not using this `home-assistant-custom-components-cover-knx-time-based` anymore since I now use `passive_addresses`.

The value restoration works using a redefined KNXCover in my xknx custom component, something like this :

```python
class KNXCover(KnxEntity, CoverEntity, RestoreEntity):
    ...
    ...
    async def async_added_to_hass(self):
        # RECOVER STATE
        await super().async_added_to_hass()
        old_state = await self.async_get_last_state()
        if old_state is None:
            return
        if (old_state.attributes.get(ATTR_CURRENT_POSITION) is None):
            return
        position = 100 - int(old_state.attributes.get(ATTR_CURRENT_POSITION))
        self._device.travelcalculator.set_position(position)

    @callback
    async def after_update_callback(self, device):
        """Call after device was updated."""
        self.async_write_ha_state()
        if self._device.is_traveling():
            self.start_auto_updater()

    @callback
    def auto_updater_hook(self, now):
        """Call for the autoupdater."""
        self.async_write_ha_state()
        if self._device.position_reached():
            self.stop_auto_updater()

        self.hass.add_job(self._device.auto_stop_if_necessary())

```
This code is inspired from my previous [custom component](https://github.com/CharlieBailly/home-assistant-custom-components-cover-knx-time-based). It basically write the state in Home Assisant when it is updated, and recover the state on Home Assistant restart using the `async_added_to_hass` callback.

There is no timeout and you are right when you say that the state could be wrong if Home Assistant is off when the state changes, but if we do not restore the state, the state will always be up (or down I can't remember) on every restarts, even if the state did not change during this restart, so I think it is always more accurate to restore the state in this situation.

`assumed_state: True` is a good idea, but I think adding state restoration can only improve this `assumed_state` functionality, don't you think ? Because yes the state could be restored wrong, but I think it is more often wrong if it always come back to a default state on each restarts.

farmio, 2020-12-05:
Actually current_cover_position on restart is `None`. Afaik It's the Frontend not accounting for this and interpreting as 0 (closed). (I don't know if there is even an issue opened for this.)

Could, in your specific case 2/4/0, move_long_address be read to request the last received movement direction from the actuator and therefore assume open/closed?

CharlieBailly, 2021-01-02:
Hi, sorry for being so long, but my knx IP Interface just died and I had to buy a new one.
But anyway, should I use xknx to make GroupValueRead on that address, or do you know any linux software I can use to achieve this (I am using [knxmap](https://github.com/takeshixx/knxmap) to monitor the bus on linux since it is very simple, but I don't think I can make GroupValueRead using this software) ?
Thanks

CharlieBailly, 2021-01-02:
Ok I find a way to read value of a certain address, but unfortunately, here is what is happening :
```txt
[ chan_id: 24, seq_no: 113, message_code: L_Data.ind, source_addr: 1.1.6, dest_addr: 2/4/0, tpci_type: UDP, tpci_seq: 0, apci_type: A_GroupValue_Read, apci_data: 0 ]
[ chan_id: 24, seq_no: 114, message_code: L_Data.ind, source_addr: 0.2.29, dest_addr: 10/4/226, tpci_type: UDP, tpci_seq: 0, apci_type: A_GroupValue_Response, apci_data: 0 ]
```
So it gives me 0 as data, but even if I change the state of the cover using one of my button, and I redo a value read, I still get 0.
Because my button is sending a payload on `10/5/66` and my ip interface on `2/4/0` and my cover is listening for both.
A solution would be to use also `10/5/66` with my interface, but the problem is I can change the cover position using another group address with another button (this other button controls multiple covers, so all the cover listen for a same address, which is neither `10/5/66` nor `2/4/0`).

farmio, 2021-01-02:
I've never heared of knxmap, but you can do GroupValueReads from xknx / HA with the `homeassistant.update_entity` service - this sends a read to all *_state_address of the entity.
But if your actuator doesn't respond properly thats no use... having no state object sucks 😕

CharlieBailly, 2021-01-03:
Yeah, having no state object sucks.
But anyway, thanks a lot for your help, I think I will just listen for knx events using an AppDaemon and set the state of the cover accordingly when I see a payload of one of my cover button on the knx bus !

closed 2021-01-03 by CharlieBailly