XKNX/xknx #525: DPT 13.013 not support float

pasarn, 2020-12-08:
**Description of problem:**

I use an MDT AMI Actor to get my energy consumption in kWh. The MDT send the value as float.

> DecodedDpValue | 1,401298E-45

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- Home-Assistant release with the issue:  0.118.5

**KNX installation:**

- AMI-1216.02 Schaltaktor mit Strommessung 12-fach, 12TE, 16/20A
- KNXD as Gateway
- Hassio on RPI

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```
knx:
  tunneling:
    host: '192.168.x.x'
    port: 3671
    local_ip: '192.168.x.x'
  sensor:
    - name: Testbrett.kwhHeizung
      state_address: '1/2/1'
      sync_state: False
      type: 'active_energy_kwh'

```
**Traceback (if applicable):**

farmio, 2020-12-08:
Well, this seems right. DPT13s are 4 Byte signed integer. 4 Byte float are DPT14 if you are looking for this.

closed 2020-12-08 by farmio

pasarn, 2020-12-08:
Hi farmio,

ETS say it can be integer (Ganzzahl) or float (Gleitkommazahl).

[image]

And the MDT AMI send a float.

[image]

[image]

farmio, 2020-12-08:
Well, maybe they should have a closer look at their own specifications. 

Nevertheless just configure a DPT14 if you want to decode 4 byte float and add the unit in HA. (Your screenshot shows 14.* as well.)

We can't implement 2 different decoders for one value_type.