XKNX/xknx #527: Strange behavior with MDT and HA

Trefex, 2020-12-12:
**Description of problem:**

- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.116.4

[image]

```
  climate:
    - name: 'Office'
      temperature_address: '2/4/0'
      target_temperature_address: '2/4/1'
      target_temperature_state_address: '2/4/3'
      temperature_step: 0.5
      min_temp: 17.0
      max_temp: 25.0
```

farmio, 2020-12-12:
> Unfortunately, it uses a 1-bit DPT so I can't use the setpoint_shift stuff.
> What am I missing here?

setpoint_shift with DPT1 is simply not supported. Use another temperature set mode if you want to use the MDT app along with HA.

Trefex, 2020-12-12:
But it shouldn’t matter the way that the set point is set? 

Why is HA using the set point value from another GA? As soon as the new set point value is sent on the Bus, HA should take that value, instead of whatever it takes? 

I edited heavily my initial post. Could you please check how I formulated it ?

Trefex, 2020-12-12:
@farmio can you explain to me in ELI5 why the fact that the app uses 1 but DPT, HA is unable to read the proper value from the GA it’s mapped to ? 

I’m trying to understand as it doesn’t make any sense currently to me.

Trefex, 2020-12-13:
It turns out that target_temperature_state_address and target_temperature_address should be the same value.

I switched the Visu app to change the comfort setpoint instead of using 1 Byte change.

It works now as intended.

closed 2020-12-13 by Trefex