XKNX/xknx #540: Instable connection to GIRA KNX IP Router 2167

stieler-it, 2020-12-21:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

I want to migrate my home automation from a rather complex Node-RED based solution to Home Assistant. I have setup Home Assistant on a Raspberry Pi 3 with Supervisor and want to migrate basically room by room. Raspberry is connected via LAN to my network and the KNX IP Router. xknx works perfect and also fast, but after a time it seems to lose it's connection. Since all our lights are controlled by KNX you can imagine how the WAF looks like currently.

What do I see? In the log there are `Heartbeat to KNX bus failed. Reconnecting.` from time to time. Maybe once in an hour.  Today there was also a `Could not connect to KNX. Retry in 3 seconds.` but I don't remember having seen that the other days. This morning our switches did not work, but as soon as I sent a command from Home Assistant this apparently fixed the connection and switches worked again. Restarting Home Assistant or waiting for a while also fixes the problem until it occurs again.

What have I tried already?
* `state_updater: false` because I thought maybe there is too much traffic or a problem with the knkx queue since a lot of my KNX devices do not react on read requests that are sent
* Switching off Node-RED so only knkx is talking to the KNX IP Router

My feeling is that the keepalive is somehow not working here, or it is not aggresive/fast enough. This doesn't really explain why the connection breaks, though. My Node-RED setup using [knx-ultimate node](https://flows.nodered.org/node/node-red-contrib-knx-ultimate) just works (I don't know if there are also reconnects, though).

**Version information:**
- xknx / Home-Assistant release with the issue: 2020.12.7
- last working xknx / Home-Assistant release (if known): Never worked for me

**KNX installation:**
* GIRA KNX IP Router 2167
* Different sensors/actors but they are all affected as soon as the connection broke.

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

```
knx:
  tunneling:
    host: '192.168.178.104'
    port: 3671
  binary_sensor: !include knx_binary_sensor.yaml
  cover: !include knx_cover.yaml
  expose: !include knx_expose.yaml
  light: !include knx_light.yaml
  sensor: !include knx_sensor.yaml
  switch: !include knx_switch.yaml
  state_updater: false

```
**Traceback (if applicable):**

`Heartbeat to KNX bus failed. Reconnecting.
Logger: xknx.log
Source: /usr/local/lib/python3.8/site-packages/xknx/io/tunnel.py:288`

Julius2342, 2020-12-21:
I have the same device. I found out, that this Gira can only handle 2 connections - when both are in use it fails to connect.

So this might happen when you have a Node-red installation running in parallel.

Try this stand-alone example to brute-force-disconnects all channels: https://github.com/XKNX/xknx/blob/master/examples/example_disconnect.py

(And i don't remember if using the device in routing mode has the same limitation - this could also be worth a try)

stieler-it, 2020-12-21:
Thanks, this is why I have disabled the KNX node in Node-RED but it didn't improve. Now I have shutdown Node-RED completely. Do you think I need to execute the script anyway?

farmio, 2020-12-21:
Hi! 
I use the same device and never had problems with it. There are different iterations of the same model (there is a sticker with the version number on it - like "I14"). Mine can handle 4 concurrent tunnels and should not close an existing one when a new connection is requested.
(Routing doesn't have these limitations as it uses Multicast - so there isn't really a connection at all.)

I think this could be a networking or power supply problem. You can also check if there is a firmware update available for the router.

If you have a newer version of the device (I think > I14 , not sure) it would support KNX-secure - xknx doesn't so this could also cause problems. But I'd think these would generate exceptions to be seen in your logs.

PS: please try to be inclusive and don't use the term "WAF".

stieler-it, 2020-12-22:
Update: Node-RED is still off, I powered off and on the KNX IP Router to get rid of possible stale connection. (For some reason I had to do that 3 times because the router seemed have some trouble starting). Home Assistant stopped the reconnect messages but KNX didn't work. So I had to restart Home Assistant also and now it's back to the old state.

Next idea: Networking and power supply should not be the problem of the KNX router (as Node-RED works without issues) but of course the Home Assistant device could have problems. I will replace all parts and try if it works better.

Thanks for making me aware of the problem with WAF. What I meant was *my* wife so the usage was incorrect and inappropriate.

stieler-it, 2020-12-22:
Firmware update of Gira KNX router did not help. I have not excluded power/networking yet, however since the Home Assistant itself has no connectivity issues... let's see.

I noticed that I have a lot of "timer_out_of_sync" events, could this be related to my KNX issue? Maybe some not-async integration that blocks the system sometimes..?

farmio, 2020-12-22:
Depending on the size of your installation and integrations your RPi may hit its limits. But that's just a wild guess. 

Have you tried to enable debug log for `xknx.log` and `xknx.knx` to see what exactly happens before the first heartbeat issue on the knx side?

Also our reconnection logic is not well tested. (And was refactored yesterday)  So let's Stick to the first connection loss after a reboot.

stieler-it, 2020-12-24:
I setup a Home Assistant installation in a virtual machine on my notebook, imported a snapshot and it worked the whole day without a single suspicious log entry. My feeling is in line with your last assumption that RPi hits the limits somehow. I will install it on the actual target hardware (custom RPi) and if I see issues again, provide you with a log. Thanks for your help so far.

farmio, 2020-12-24:
Would be interesting if the knx integration was the one using up too much resources. Can you maybe try, on the RPi, to just disable knx and see if there are still timer_out_of_sync events?

stieler-it, 2020-12-25:
I moved the whole setup to the live hardware now (also Raspberry like) but still cannot reproduce the problems any more. Although I have enabled Node-RED with KNX again for the use cases I have not yet migrated to Home Assistant. So, no clue what happened and I will close this ticket for now. Thanks again for your support.

closed 2020-12-25 by stieler-it