XKNX/xknx #548: Home Assistant missing telegrams (tunneling connection breaking ?)

kalhimeo, 2021-01-01:
**Description of problem:**

I am still digging into the issue, but I have noticed that my Home Assistant is sometimes missing some KNX telegrams. I cannot say exactly what causes the problem so far because I only notice the problem later when important telegrams have not been received by HA. I have only found this error in logs (see below) which is maybe linked to the problem.
My feeling is that something breaks the tunneling connection which would explain why I miss a few telegrams sent during a particular time period, before the connection recovers itself (but I may be wrong, as I said I am still digging into the exact cause and how to reproduce). What I am sure is that the telegrams which are not received by HA were well sent on the KNX bus because I have other KNX devices monitoring the same GA and which show the proper value.

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 2020.12.2
- last working xknx / Home-Assistant release (if known): 0.117 was working perfectly. 0.118 had other problems with KNX state_updater so I am not sure if that version was 100% working or may already had this bug.

**KNX installation:**
Tunneling via MDT IP Gateway

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
configuration.yaml

**Traceback (if applicable):**

> Logger: homeassistant
> Source: components/knx/__init__.py:441
> First occurred: 30 December 2020, 19:43:59 (6 occurrences)
> Last logged: 1:54:18
> 
> Error doing job: Task exception was never retrieved
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.8/site-packages/xknx/dpt/dpt_scaling.py", line 45, in to_knx
>     raise ValueError
> ValueError
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/components/knx/__init__.py", line 424, in _async_entity_changed
>     await self._async_set_knx_value(new_attribute)
>   File "/usr/src/homeassistant/homeassistant/components/knx/__init__.py", line 441, in _async_set_knx_value
>     await self.device.set(value)
>   File "/usr/local/lib/python3.8/site-packages/xknx/devices/expose_sensor.py", line 70, in set
>     await self.sensor_value.set(value)
>   File "/usr/local/lib/python3.8/site-packages/xknx/remote_value/remote_value.py", line 167, in set
>     payload = self.to_knx(value)  # pylint: disable=assignment-from-no-return
>   File "/usr/local/lib/python3.8/site-packages/xknx/remote_value/remote_value_sensor.py", line 58, in to_knx
>     return DPTArray(self.dpt_class.to_knx(value))
>   File "/usr/local/lib/python3.8/site-packages/xknx/dpt/dpt_scaling.py", line 51, in to_knx
>     raise ConversionError("Could not serialize %s" % cls.__name__, value=value)
> xknx.exceptions.exception.ConversionError: <ConversionError description="Could not serialize DPTScaling" value="102"/>

farmio, 2021-01-01:
Hi!
Seems you are trying to use an ExposeSensor with an invalid payload. 
Can you show some configuration of the Entities not working/causing errors?

PS: please use code-Blocks (backticks) instead of quotes for logs and code to preserve spacing and special characters and for better readability.

kalhimeo, 2021-01-01:
Hi farmio, thanks for the super fast feedback :-) Sorry for the quotes, will use code-blocks next time.

I have just checked all my exposed sensors, all seem to have the right "type" according to the intended payload. I can post the config of all my exposed entities if that's what you are interested in ? Also could this error break the tunneling connection ?

farmio, 2021-01-01:
You may want to enable debug logging for "xknx.log" and "xknx.knx" (via the HA configuration or service) to figure out what is causing this problem.

kalhimeo, 2021-01-01:
Alright, I just enabled debug in configuration, will try to spot the moment when things get broken. Thanks for your help !

kalhimeo, 2021-01-05:
After a new occurrence of the problem, there are good chances that it does not come from the KNX integration. It appears from logs that my whole HA installation is hanging which results in the KNX tunneling to time out (but other integrations as well...). I initially thought that only the KNX connection got lost as it was the only one "leaving traces" in my setup, but as I was able to face the problem in live, I could see that HA was not reachable at all even by web browser during the 10-20min freeze.
I am still looking for the exact cause of these lags, but as it probably comes from an external factor, I am closing this issue.
Thanks for your help !

closed 2021-01-05 by kalhimeo