XKNX/xknx #561: Way to read and write existing rules

Chrescht, 2021-01-12:
Hi all,

this is more a question than an issue.

I am quite new to this so please bear with me.
Yesterday night I was quite happy when I was able to finally switch a light on and off using xknx.

Now when the system was installed, they added rules. One example is to open the shades if there is a storm blowing (wind speed over threshold for a given amount of time).
Is there a way to access those rules and to modify them or would I have to use another tool e.g. @knxd ?
Is it correct that that'd be called _ETS programming_?

many thanks

farmio, 2021-01-12:
I think the "rules" you refer to are functions of your actuators. They live in your knx devices (blind actuator, switch, etc) - decentralized. They are configured with the Windows software ETS. To change configuration you need the .knxproj file of your installation - without it you would need to configure the whole knx installation new from scratch. 
This has nothing to do with xknx or knxd. These are just protocol implementations.

Chrescht, 2021-01-12:
Ok thanks for the clarifications, I was afraid of such an answer.

I take it modifying devices is part of the ETS configuration too e.g. modifying a switch button.

So if I now was to configure my own logic using xknx, then both systems would function at the same time, and potentially work "against each other" which would not be ideal.

At least it's possible to add some logic using xknx which solves part of my problem (== not having to call the electric company).

farmio, 2021-01-12:
> modifying devices is part of the ETS configuration too e.g. modifying a switch button.

Right.

> both systems would function at the same time, and potentially work "against each other"

This may or may not be the case. However I'd call the electric company to get the knxproj file anyway ;)
If you speak german https://knx-user-forum.de is a good source for generic knx questions.

closed 2021-01-12 by farmio