XKNX/xknx #583: Feature request: Rounding of sensor values according to KNX specificaiton

KNXBroker, 2021-01-29:
XKNX sensors are sometimes displayed with lots of decimals, which do not make sense. Example: power = 21.123123123W

Sensors values should be rounded by XKNX according to the KNX standard. Example 0 decimals for DPT14.00 [image]
For detailed specification see KNX System 3 Specifications -> 7 Interworking ->2 Datapoint Types

My proposal is to add an attribute resolution to each applicable dpt class (https://github.com/XKNX/xknx/blob/master/xknx/dpt).

```
class DPTAcceleration(DPT4ByteFloat):
    """DPT 14.000 DPT_Value_Acceleration (ms-2)."""

    dpt_main_number = 14
    dpt_sub_number = 0
    value_type = "acceleration"
    unit = "m/s²"
    resolution = 0
```

This attribute could be used afterward in https://github.com/XKNX/xknx/blob/master/xknx/devices/sensor.py for rounding.

Commit https://github.com/XKNX/xknx/pull/304/commits/7687ef3f818950240e96a96979bdeb78b86cac23 would not be required.

Thanks for your comments

farmio, 2021-01-29:
👍 
I think the rounding should even be done in the DPT class instead of Sensor.

KNXBroker, 2021-02-04:
Hello, I have tried too understand XKNX for preparing a PR, but I have soon recognized that my coding skills are not good enough.

First of all I have recognized that the attribute `resolution` does already exist in the DPT classes, however they are handeled differently. They seem to be correct in every DPT class, expect for `DPT2ByteFloat(DPTBase)`. In this class (which is defining DPT9) resolution is always 1, however according to the KNX specification it should be 0.01. My assumption is that the class `from_KNX` is using the resultion differently for `DPT2ByteFloat`. However I could not find out how this is exactly done.

This should be harmonized at the very beginning. Afterwards `round()` function could be added to all `from_KNX` classes, taking the `resolution` attribute as reference (1 by default).

If someone else wants to prepare a PR, you are welcome 😃 

Best regards, KNXBroker

farmio, 2021-02-04:
resolution is just not used anywhere 😉
However I have tested `DPT2ByteFloat.from_knx()` with every value and it is returning exactly the same as ETS would display. This is not true for `DPT2ByteFloat.to_knx()` but I haven't looked into this yet.

4byte floats are encoded/decoded very differently. These should be rounded properly.

KNXBroker, 2021-02-04:
> resolution is just not used anywhere 😉

This explains a lot, maybe my skills are better than expected 😄 

In specific, I have recognized the issue with `class DPT4ByteFloat(DPTBase)`.
I'm using DPT 14.056 | power in one of my installations and the value is something like this: 32.1231235431
According to KNX specification DPT14 should not have any decimals.

KNXBroker, 2021-02-04:
If it helps, I could check raw data with ets (it's an installation of a friend, the number above was just an example)

farmio, 2021-02-04:
It doesn't need an installation, apart from installing ETS on Windows. In Group Monitor you can send any number or even hex value.

KNXBroker, 2021-02-04:
That was easy. I have tested 14.056
- numbers without decimals are working perfect (e.g. 4 = 4)
- numbers with decimals are showing the crazy numbers (e.g. 4.4 = 4.400000095367432)

Conversion is correct.

Resolution of 14.056 according to KNX specification: 1W
Interesting: Resolution does not mean number of decimals!! According to ETS and according to the actuator's manual decimals are allowed for DPT1.

As a next step I have tested 4.123456789
- in ETS it is display as 4,123457 --> rounding with 6 decimals
- in Home Assistant it is display as 4.123456954956055 --> incorrect after 6 decimals

I would propose rounding of DPT14 (IEEE 754 formated) to 6 decimals (as ETS does).  This should avoid crazy numbers in Home Assistant. 

What do you think?

KNXBroker, 2021-02-05:
Just rechecked the existing commit #7687ef3, that ist exactely what it does. I'm wondering about rounding to 7 decimals and I'm wondering about rounding 2byte floats.

@smurfix , May I ask you to provide some additional information? Was there any reason why you have stopped working on it (except of lack of time...).

farmio, 2021-02-05:
I think the encoding should be reversible. But I'm not even sure if this is possible without rounding.

You can test it.
```python
# Pseudocode
for x in range(0xFFFF):
    x == DPT2ByteFloat.from_knx(
            DPT2ByteFloat.from_knx(x)
        )
```

smurfix, 2021-02-05:
It's always possible to revert a float encoding, at least unless you have multiple non-numeric values (NaN or Inf).

farmio, 2021-04-08:
> Sensors values should be rounded by XKNX according to the KNX standard

There is no more resolution column for DPT14 in the current version of this document (v02.01.02).

> 3.15 Datapoint Types “4-Octet Float Value”
The resolution is given by the use of the IEEE 754 format and varies with the used exponent.

KNXBroker, 2021-04-09:
> Interesting: Resolution does not mean number of decimals!!

Resolution does not mean rounding, this was a misunderstanding at the very beginning from my side, sorry for confusion. 

> I would propose rounding of DPT14 (IEEE 754 formated) to 6 decimals (as ETS does). This should avoid crazy numbers in Home Assistant.

I think the approach of @smurfix was / is correct.

farmio, 2021-04-09:
What does it mean then? 
I think it may have meant exactly that, but the specs changed for whatever reason and now it is not defined anymore. See https://www.knx.org/wAssets/docs/downloads/Certification/Interworking-Datapoint-types/03_07_02-Datapoint-Types-v02.01.02-AS.pdf
<hr>

I don't understand smurfixs calculation:
```python
res = struct.unpack(">f", bytes(raw))[0]
res = round(res, 6-int(log10(abs(res))))
return res
```
what is the difference to using `round(6)`?

smurfix, 2021-04-09:
The difference is that floating point numbers have a number of significant digits (~7 in this case) *in total*, not just behind the decimal point. It cannot transmit the difference between 1000 and 1000.0001, so this formula rounds the fake accurracy off at the third digit.

The problem with DPT14 that this formula solves is that in practice many devices take a decimal float (as in, the "human-readable", already-rounded-off value that it wants displayed on your nice and expensive KNX visualization gadget) with *at most* five significant digits, converts that to a 32-bit float, and sends it via DPT14. My formula always exactly reproduces the original value while not losing too much accuracy (which is mostly fake anyway, nobody cares about the 6th digit on a device that has 2% error, or possibly 0.1% if you calibrate it) if the sender should happen to start off with a "real" float.

farmio, 2021-04-11:
@smurfix 
If you are still using this in your fork, consider escaping the value `0` as `log10(0.0)` would raise a ValueError.
Also `int(0.x) == int(-0.x)` so its off by one for values < 1
```python
>>> from math import log10
>>> int(log10(0.03))
-1
>>> int(log10(0.3))
0
>>> int(log10(3.3))
0
>>> int(log10(34.3))
1
```
I have settled with
```python
# prevent log10(0.0) -> ValueError: math domain error
if abs_float := abs(raw_float):
    # round to 7 digit precicion independent of exponent - same value as ETS 5.7 group monitor
    return round(raw_float, 7 - ceil(log10(abs_float)))
return 0.0
```
now.

smurfix, 2021-04-11:
Thanks for the correction.

PR #667 (2021-04-11): round DPT 14 values

## Description

Round DPT 14 values the way ETS does (presumably 🙃).

Test values are checked with ETS group monitor.

closes #583 

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

## Checklist:

- [ ] I have performed a self-review of my own code
- [ ] The documentation has been adjusted accordingly
- [ ] The changes generate no  [... cut]
closed 2021-04-11 by marvin-w