XKNX/xknx #603: Proper handling of old cover KNX devices like TR521 - Hager

Tolriq, 2021-02-14:
Hi,

So I've read a lot of messages about such devices that does not expose a position in percent. But it seems there's no final solution documented.
I still have 5 of them so would love to have them working properly.
Those devices only expose a DPT 1.001 as readable state that shows the last direction used and properly notify the bus when using the physical buttons.
Unfortunately it seems this is not supported at all currently so we can't use the state_address.

While I could eventually force everyone to use HA and not the physical buttons to control those, there's still the issue that the position is not saved on HA reboot and it seems there's no proper official way to save state those for covers.

Is there anything I can do to have them better integrated?

farmio, 2021-02-14:
The button presses should be recognized by HA if you set the move_long address accordingly. So this already is on the Bus - no need for a status here. 

The state after reboot is undefined in HA so you should be able to use both UP or DOWN button. This is until one is used or a telegram on the move_long address (from physical button) is received. 

Except for reading the status on HA startup, is there anything not working? 
Saving can cause troubles when the covers moved since HA was shut down (down button disabled while covers are open...) so we went for undefined.

Tolriq, 2021-02-14:
Not sure to follow.

Those devices have physical buttons connected to them to control them, those entries are in proper move long group address but no message is sent on the bus at all (Those are radio devices).

The only messages send on the bus are from the state switch value but it's DPT 1.001 so can't be used as state address since 5.x is required for that.

farmio, 2021-02-14:
Ah ok now I get it. Can you set the state sent from the actuator to the same GA that will trigger it, or would this cause troubles?

As a workaround you can customize the HA entity with `assumed_state = true`. 
Or use the state in a logic / automation to send a fake position in % to a dummy GA. 

We are looking into supporting passive group addresses in the future. This would probably solve your problem with just a little patience (months with current HA review times...).

Tolriq, 2021-02-14:
From a quick test they do not like the state sent to the group it loops events and breaks it :(

> Or use the state in a logic / automation to send a fake position in % to a dummy GA.

Do you have more details about how to handle that properly? Because if I add a faked state group, HA will stop guessing the value from the durations for all the usages from HA, that would not be cool. Since the state message is always sent by the actual actuator, there's no way to differentiate who triggered the action to eventually use the calculated value from the cover object.

The main final problem is that those are a little dumb and even if fully opened sending an open command will run the motor for another full open duration and it's bad for the motors and fixation system :(
So the state is important for me for all the automation parts like auto closing / opening on events that should not run if the state is already the targeted one :( So the assumed_state fix the UI to manually do things, but not the automation part :(

Tolriq, 2021-02-16:
@farmio While searching for a solution I found that state can be updated via a python script.

If I use that to set the current_position of the cover is there side effects (those have no position address so can't have position changed via ui) and will the UI / xknx internals properly update without issues ?

farmio, 2021-02-16:
If you don't set a position_address but only a (fake) position_state_address you should be fine.

Tolriq, 2021-02-16:
@farmio ok it works nicely even without putting a fake position_state_address that's nice

I think I can automate :
- when status unset read the switch value and put a fake position.
- when switch state change wait X second if status is fully open but last switch says close or opposite then buttons where used, put fake positions.

Won't be perfect if someone stop the covers in the middle from the physical button (can probably handle this too with time diff between the events and calculation but way more complicated :() or with a restart of the server when half position but it's still a lot better.

The only small problem is the X, it seems the covers does not expose the configured travelling_time durations so I'll have to hard code that. Do you think this could be exposed in the future as state on the cover from xknx ? Or are you aware of another way to get this value?

farmio, 2021-02-16:
> I found that state can be updated via a python script

depending on what exactly you are doing here the answer to your questions could vary. 
I have no idea if this works, but you could maybe inject your state-GA to `KNXCover._device.updown.passive_group_addresses`. This would expect a type `List[xknx.telegram.GroupAddress]` which you can get from calling `[xknx.telegram.GroupAddress("group_address_string")]`
Good luck!

Tolriq, 2021-02-16:
@farmio sounds way too complicated for me :)

My scripts and hacks covers most of my needs even if not perfect. The only issue left is that I have no clue about how to get the value of 'travelling_time_down' and 'travelling_time_up' from inside HA.
With that last piece I can easily automate the fixing with just the cover entity ID and a fake switch entity ID that points to that state GA.
Would really appreciate some idea / solution for that last part.

farmio, 2021-02-16:
I'm confused. This is in no way more complicated than writing a python script to circumvent it. I still have no idea how and where you run you python script / hack so I don't know what you have access to. 
If you have access to the `hass` object you could probably extract it form `hass.data["knx"].config` somewhere or `hass.data["knx"].xknx.devices["<device_name>"].travel_time_down` or so but then again I'd think about my previous suggestion.

Tolriq, 2021-02-16:
I was maybe not clear I use a found python script that allows updating any state of any entity and just call it with proper params.

I hate python so would rather avoid building and maintaining something in python, but will if I have no choice.

My hack is currently to define KNX switch entities pointing to the GA state address:
```
- name: 'Cover state XXXXX'
  address: '1/1/1'
  state_address: '1/1/1'
```

With those entities I can react to events and get last state to automate everything I need, when real cover load first to unknown state, get the state of the fake switch and set cover position / state from it.
When event on fake switch wait for the configured duration (that I'd like to get automatically) and if real state does not match fake switch test then it was physical button and update real cover state with guessed values.

This covers state restoration from the actual covers (approximately) and physical button press so all my needs except when covers are stopped mid course, but that's not that important for my usage.

So with easy reading of the durations I can do most without touching python.

Tolriq, 2021-02-20:
@farmio So went the python way and got nearly everything working.

My only remaining issue is how to properly restore the xknx state after a server restart.

I can easily fix the HA side of the state to have the UI buttons ok, but xknx internal states (travel / remotevalue) are not updated so the first UI usage after a restore triggers some small side issues.

Is there a way to update those too?

farmio, 2021-02-20:
https://github.com/home-assistant/core/pull/46670

Tolriq, 2021-02-20:
Thanks.

So if anyone comes here, here's a simple yet relatively proper solution using pyscript.
Create switch entities with the state address of the binary position of the cover, use customisation to add a custom attribute associated_cover that points to the switch identity. And pass the list of switchs.

```
from xknx.telegram import GroupAddress, Telegram
from xknx.telegram.apci import GroupValueRead

@time_trigger("startup", kwargs={'switches' : [ 'switch.etat_store_salon_terrasse', 'switch.etat_store_cuisine', 'switch.etat_store_bureau', 'switch.etat_store_salon_cuisine', 'switch.etat_store_salon_rue']})
def fix_covers_from_switches(**kwargs):
    switches = kwargs.get('switches')
    xknx = hass.data['knx'].xknx
    for switch_entity_id in switches:
        switch_attributes = state.getattr(switch_entity_id)
        if switch_attributes == None:
            log.warning(f"Unknown switch entity {switch_entity_id}")
            return
        cover_entity_id = switch_attributes.get('associated_cover', None)
        if cover_entity_id == None:
            log.warning(f"No associated cover {switch_attributes}")
            return
        cover_attributes = state.getattr(cover_entity_id)
        if cover_attributes == None:
            log.warning(f"Unknown cover entity {cover_entity_id}")
            return
        switch_device = xknx.devices[switch_attributes['friendly_name']]
        cover_device = xknx.devices[cover_attributes['friendly_name']]
        cover_device.updown.passive_group_addresses = [GroupAddress(switch_device.switch.group_address_state)]
        i = 0
        while(state.get(cover_entity_id) == "unknown" and i < 3):
            xknx.telegrams.put(Telegram(destination_address=GroupAddress(switch_device.switch.group_address_state), payload=GroupValueRead()))
            task.sleep(3)
            i = i + 1
        log.warning(f"Hack applied: {switch_entity_id} <-> {cover_entity_id}")

```

farmio, 2021-02-20:
Why do you need the switch entity? I think you could just apply the passive group address to the cover and be done.

Tolriq, 2021-02-20:
Hum yes I can customize the cover to add the GA as attribute and drop it.
I started like that because I used that to get the state after restart before the mandatory switch to python and direct access to xknx.

Tolriq, 2021-02-20:
@farmio Updated the script to be a lot better and simpler.

Can I assume that when passive group address are merged in HA they will be automatically queried on startup too?

farmio, 2021-02-20:
No, they won't. Thats why they are passive. Only `*_state_address`es are automatically queried.

Tolriq, 2021-02-20:
Ok thanks, so I'll have to keep this hack around, good to know.
I guess this can be closed then.

closed 2021-02-20 by Tolriq

farmio, 2021-02-20:
It's just the configuration thats missing to add GAs to passive_group_address for the feature to exist. When this is done you can configure your GA as passive of move_long and just add an automation to HA that calls the knx.read service for these GAs on startup.

Tolriq, 2021-02-20:
Ok thanks for the help even if unclear at start :)