XKNX/xknx #615: Better handling of dumb KNX covers

Tolriq, 2021-02-25:
So me again for the last small wanted improvement to my handling of those stupid dumb old KNX cover devices :(

Those device despite having the durations in their ETS config do not use them properly to calculate half states.

So when the covers are stopped in half positions, and a long move command is sent the motor will run for the whole duration and not for the proper half duration.
Since the position and timing are correct inside xknx I wonder if there's a way to natively have the library send a stop message when the cover reach 100% or 0% in those case?

If not what would be the best way to detect that's it's indeed a move from a mid position to only send the stop in those cases?

farmio, 2021-02-25:
> when the covers are stopped in half positions, and a long move command is sent the motor will run for the whole duration

This is, depending on the actuator, expected behavior. The position calculated from time is not accurate and even less so when doing multiple travels without reaching an end-stop (eg. 10-20-30-40-50... % over the day). The error adds up. 
So for the actuator to be sure the position is correct it needs a reference-run - doing a move_long for the full movement time is such. 
This happens without notice because the motor stops from end-stop-switches or internal electronics. It's just the actuators Relais that is on for some seconds longer.

Tolriq, 2021-02-25:
The problem is that all that installation is from previous owner and the motors are bubendorf 4 wires so in this case don't stop forcing when reaching the limits.

And it's not some time some seconds it's every time the whole duration :( So if someone by mistake press down and stop one second later. The next up will run for 18 seconds. 

This already broke 2 motor to cover fixations + 3 small part that prevent the store to go after the top. One of the expected side effect to move to HA was to have more control over such things to reduce the issue.

Seems there's no normally priced solution for those motors, and changing all motors + actuators is well out of budget.

I'll guess I'll send the stop each time 1/2 sec after a transition from x to 100 or x to 0. And manually adjust the covers when the position error offset is too big.

farmio, 2021-02-25:
You can easily do this with a HA automation.

Tolriq, 2021-02-25:
Yes no problem on that part, just wanted to be sure I was not missing something obvious like the passive group the other time :)

Thanks again for the details.

closed 2021-02-25 by Tolriq