XKNX/xknx #621: lights state is not updating regularly

adamr2, 2021-03-02:
**Description of problem:**
light state is not updating regularly when changing it's state (on\off) from outside of HA. State only updates once in ~10 state changes.

- [ ] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: latest

**KNX installation:**
[image]

Configuration:
```
knx:
  tunneling:
    host: "10.0.0.70"
    port: 50000
    local_ip: "10.0.0.10"
  light:
    - name: "Hila"
      address: "0/3/17"
      state_address: "7/3/17"
```

Logs:
```
2021-03-02 21:06:18 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="CONNECT_REQUEST" Reserve="0" TotalLength="26" />
body="<ConnectRequest control_endpoint="<HPAI 10.0.0.10:37404 />" data_endpoint="<HPAI 10.0.0.10:37404 />" request_type="ConnectRequestType.TUNNEL_CONNECTION" />" />
2021-03-02 21:06:18 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061002060014100008010a000046c3500404117f
2021-03-02 21:06:18 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="CONNECT_RESPONSE" Reserve="0" TotalLength="20" />
body="<ConnectResponse communication_channel="16" status_code="ErrorCode.E_NO_ERROR" control_endpoint="<HPAI 10.0.0.70:50000 />" request_type="ConnectRequestType.TUNNEL_CONNECTION" identifier="4479" />" />
2021-03-02 21:06:18 DEBUG (MainThread) [xknx.log] Tunnel established communication_channel=16, id=4479
2021-03-02 21:06:18 INFO (MainThread) [xknx.log] Successfully reconnected to KNX bus.
2021-03-02 21:06:24 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200015041000002e00bce0117f570c010080
2021-03-02 21:06:24 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="TUNNELLING_REQUEST" Reserve="0" TotalLength="21" />
body="<TunnellingRequest communication_channel_id="16" sequence_counter="0" cemi="<CEMIFrame SourceAddress="IndividualAddress("1.1.127")" DestinationAddress="GroupAddress("10/7/12")" Flags="1011110011100000" payload="<GroupValueWrite value="<DPTBinary value="0" />" />" />" />" />
2021-03-02 21:06:24 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="TUNNELLING_ACK" Reserve="0" TotalLength="10" />
body="<TunnellingAck communication_channel_id="16" sequence_counter="0" status_code="ErrorCode.E_NO_ERROR" />" />
2021-03-02 21:06:24 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:24 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:25 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:25 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:26 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:26 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:27 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:27 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:28 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:28 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:29 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:29 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:30 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:30 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:31 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:31 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:32 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:32 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:33 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:33 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:34 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:34 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:35 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:35 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041001002e009de000000000000000ff
2021-03-02 21:06:35 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061002090010100008010a000046c350
2021-03-02 21:06:35 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="DISCONNECT_REQUEST" Reserve="0" TotalLength="16" />
body="<DisconnectRequest CommunicationChannelID="16" control_endpoint="<HPAI 10.0.0.70:50000 />" />" />
2021-03-02 21:06:35 WARNING (MainThread) [xknx.log] Received DisconnectRequest from tunnelling sever.
2021-03-02 21:06:35 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="DISCONNECT_RESPONSE" Reserve="0" TotalLength="8" />
body="<DisconnectResponse CommunicationChannelID="16" status_code="ErrorCode.E_NO_ERROR" />" />
2021-03-02 21:06:35 DEBUG (MainThread) [xknx.log] Closing transport.
2021-03-02 21:06:38 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="CONNECT_REQUEST" Reserve="0" TotalLength="26" />
[... 356 lines cut ...]
2021-03-02 21:09:25 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="DISCONNECT_REQUEST" Reserve="0" TotalLength="16" />
body="<DisconnectRequest CommunicationChannelID="24" control_endpoint="<HPAI 10.0.0.70:50000 />" />" />
2021-03-02 21:09:25 WARNING (MainThread) [xknx.log] Received DisconnectRequest from tunnelling sever.
2021-03-02 21:09:25 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="DISCONNECT_RESPONSE" Reserve="0" TotalLength="8" />
body="<DisconnectResponse CommunicationChannelID="24" status_code="ErrorCode.E_NO_ERROR" />" />
2021-03-02 21:09:25 DEBUG (MainThread) [xknx.log] Closing transport.
2021-03-02 21:09:28 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="CONNECT_REQUEST" Reserve="0" TotalLength="26" />
[... cut ...]
2021-03-02 21:09:30 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041901002e009de000000000000000ff
2021-03-02 21:09:31 DEBUG (MainThread) [xknx.raw_socket] Received from ('10.0.0.70', 50000): 061004200014041901002e009de000000000000000ff
2021-03-02 21:09:31 DEBUG (MainThread) [xknx.knx] Unsupported KNXIPFrame: APDU LEN should be 0 but is 3 in CEMI: 2e009de000000000000000ff in 061004200014041901002e009de000000000000000ff
```

farmio, 2021-03-02:
Your connection is constantly being closed by the server after sending some unsupported frames. I guess the lights don't update while reconnecting / waiting for reconnect.
What device are you connecting to and what are these unsupported frames?

adamr2, 2021-03-02:
MS3-2R Maestro Server by CD Innovation. I don't know what these frames are. 
The server has KNXnetIP on port 50000 that works with ETS3 only, supports tunneling only and dos not support routing

farmio, 2021-03-07:
It seems your interface is sending invalid  L_DATA_CON confirmation frames.
I'd talk to the manufacturer, maybe there is already an update available. 

Since you are using xknx stand-alone you could try to comment out this line in xknx to disable parsing L_DATA_CON completely 
https://github.com/XKNX/xknx/blob/73495a2135df99a2143fc15fb406ed406b3696b0/xknx/knxip/cemi_frame.py#L125
and see if it works then. The confirmation frames are currently not used in xknx, but parsed nevertheless and this may not fail.

closed 2021-04-08 by farmio