XKNX/xknx #622: Must always sleep after every interaction with the bus

tails4e, 2021-03-02:
**Description of problem:**
Every time I issue a command to the bus it works, but only if I use "await asyncio.sleep(2)" before the next command - 
i.e. if I do:
     temp  = xknx.devices[1]
     await temp.sync() # Syncs the state. Tries to read the corresponding value from the bus.
     print(temp.resolve_state())

I get a temperature of None - but if I sleep before it works.
     temp  = xknx.devices[1]
     await temp.sync() # Syncs the state. Tries to read the corresponding value from the bus.
     await asyncio.sleep(2)
     print(temp.resolve_state())

Similarly if I try to turn on 2 lights without having a sleep between them, it fails. I don't mind working around this by adding a delay to every call, but it seems like this should be unnecessary

- [ x] using xknx standalone
- [ ] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0,17
- last working xknx / Home-Assistant release (if known): NA

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
general:
  own_address: "1.1.120"

connection:
  tunneling:
    gateway_ip:   "192.168.1.149"
    gateway_port: "3671"
    local_ip:     "192.168.1.167"

groups:
  light:
    Living-Room.Light_1: { group_address_switch: "1/1/16" }

  # Cyclic sending of time to the KNX bus
  time:
    General.Time: { group_address: "1/1/100" }

  sensor:
    Cinema.Temperature: {group_address_state: '1/1/101', value_type: 'temperature'}

**Traceback (if applicable):**

farmio, 2021-03-03:
Use `sync(wait_for_result=True)`
https://github.com/XKNX/xknx/blob/a8ff83416c5496f64fe62bc2b0ab31c373a40cf9/xknx/devices/device.py#L77
to wait for an answer. Without it just sends the GroupReadRequest. 

This doesn't apply to turn_on. This should work as expected. (If you had configured 2 lights in your configuration)

marvin-w, 2021-03-05:
Also please keep in mind that we will no longer support the xknx.yaml in the future, so you might want to switch to manually creating your devices.

farmio, 2021-03-07:
Updated docs to make this clearer #624

tails4e, 2021-03-07:
> Also please keep in mind that we will no longer support the xknx.yaml in the future, so you might want to switch to manually creating your devices.

That's a shame, the YAML seems quite useful. what's the rationale for removing it, and what is the recommended way for managing lots of devices after this?

farmio, 2021-03-07:
It depends on what you want to do with it. You can just define the devices in plain Python. 

We figured most applications using xknx will probably need their own configuration Schema since just defining xknx devices is not particularly useful - you will want to assign them to your application as well.

closed 2021-03-14 by farmio