XKNX/xknx #629: FAQs

farmio, 2021-03-12:
Add FAQs to the documentation to handle common questions and issues. 

- How to debug `[xknx.log] Error: KNX bus did not respond in time (2.0 secs) to GroupValueRead request for / [xknx.log] Could not sync group address `
- multiple installations
- cover: inverted % in HA
- received DisconnectRequest

Tolriq, 2021-03-27:
Hey there, do you have any timeline about the debug stuff and the could not sync?

I'm having issues with sometimes a random cover not acting when doing operation on a cover group and would like to see if there's something obvious and I should reduce even more the rate_limit.

farmio, 2021-03-27:
No there is no timeline for any work on this project. 
For debugging tips you can currently see the sticky posts in the discord servers general channel or in the HA community forum. 

PS You should probably not reduce rate_limit, but this is for another issue.

closed 2021-11-25 by github-actions[bot]