XKNX/xknx #669: unstable connect to eibport3, see issue #55

teampyt, 2021-04-13:
Hi, I use a raspberry PI 4 and the latest OS build, python3.9.4 and today's xkxnx.
I test with the example gateway-scan and get the following responses...

pi@raspberrypi:~ $ python3 gwscn.py
No Gateways found
pi@raspberrypi:~ $ python3 gwscn.py
No Gateways found
pi@raspberrypi:~ $ python3 gwscn.py
No Gateways found
pi@raspberrypi:~ $ python3 gwscn.py
Gateway found: AppModule-1 KNXNet/IP / 192.168.210.235:3671
- Device supports tunneling
- Device supports routing, connecting via 127.0.0.1
Gateway found: AppModule-1 KNXNet/IP / 192.168.210.235:3671
- Device supports tunneling
- Device supports routing, connecting via 192.168.210.91
Gateway found: eibPort-3 KNXNet/IP / 192.168.210.234:3671
- Device supports tunneling
- Device supports routing, connecting via 127.0.0.1
Gateway found: eibPort-3 KNXNet/IP / 192.168.210.234:3671
- Device supports tunneling
- Device supports routing, connecting via 192.168.210.91
pi@raspberrypi:~ $

As well, any application will work only one in a while. If the gateway is found, all is well...
What am I doing wrong?

thanks for your help!
teamvcpyt

farmio, 2021-04-14:
Hi! 
- Have you tried using Wireshark to capture what is sent/received on the network when the gateway is found vs. not found?
- Have you tried set logging in xknx to `debug`?

teampyt, 2021-05-04:
Hi!
Sorry for my late response....
I am not familiar with wireshark, seems to be a mighty tool.
No, I will work my way through and get the xknx debug running. Any hint where to read?
I am just getting started on the raspberry, so maybe I will need a bit of additional help. ;-)

Maybe someone can post a link on an example that uses direct IP addresses instead of the gateway discovery? This might get me up and running.
Thanks!

farmio, 2021-05-04:
Maybe my little testscript can get you started

```python
import asyncio
import logging
import time

from xknx import XKNX
from xknx.devices import Sensor
from xknx.io import ConnectionConfig, ConnectionType

logging.basicConfig(level=logging.INFO)
logging.getLogger("xknx.log").level = logging.DEBUG
logging.getLogger("xknx.knx").level = logging.DEBUG

async def main():
    connection_config = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING,
        gateway_ip="10.1.0.40",
    )
    xknx = XKNX(connection_config=connection_config)

    sensor = Sensor(
        xknx,
        name="Sensor",
        group_address_state="1/2/3",
        value_type="temperature",
    )

    async with xknx:
        start_time = time.time()
        await sensor.sync(wait_for_result=True)
        print(
            f"Value: {sensor.resolve_state()} - took {(time.time() - start_time):0.3f} seconds"
        )

asyncio.run(main())
```

teampyt, 2021-05-07:
Dankeschön Matthias,

mit der Angabe der IP funktioniert es. Nun kann ich versuchen damit laufen zu lernen...
Dann suche ich mal nach Beispielen. Vielleicht gibt es ja was?
Ich suche eine Möglichkeit einen knx statetable zuvor definierter Gruppenadressen im Speicher zu haben, dort einzelne Werte zu änden und dann diese Änderungen auf den Bus auszugeben. Wenn jemand so ein Modul schon mal gemacht hat, dann wäre ich sehr dankbar dies als Ausgangspunkt für mein Projekt (Dynamische Steuerung von Solarenergieverbrauchern) zu realisieren.
VG
teamvc

closed 2021-11-13 by farmio