XKNX/xknx #693: Home-Assistant knx plugin locks KNX IP Interface completely

Blade0001, 2021-05-11:
I am not sure but it seems that since the last Homeassistant updates the knx plugin blocks by interface completely. 
When Homeassistant is running it not possible to use the interface for ETS as well. 
When I restart or shut down home assistant the interface is free to use for ETS. 

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:
- Home-Assistant release: 2021.5.1

**KNX installation:**

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
knx:
  tunneling:
    host: '***-***-***.101'
    port: 3671
    local_ip: '***.***.***.32'

**Traceback (if applicable):**

farmio, 2021-05-12:
HA 2021.5.2 seems to work fine on my installation.

@Blade0001 please try to share as much information as possible, not as little. Good sources to debug are
- xknx logs (especially `xknx.log` and `xknx.knx` in this case)
- ETS Busmonitor logs
- Wireshark captures

from working installation / starting HA until the problem starts.
Don't worry about [private IP address range](https://en.wikipedia.org/wiki/Private_network), this is not critical.

Blade0001, 2021-05-12:
HI Farmio, 

where I can find these logs? Please find my HA xknx.log attached. 
The ETS logs say nothing. Because when I run the KNX monitoring homeassistant is not able to connect to the interface and vice versa. You can see this in the logs. 
[home-assistant.log](https://github.com/XKNX/xknx/files/6465684/home-assistant.log)
`

Blade0001, 2021-05-12:
I have tried one thing. I have also the Node-RED plugin installed in HA. There I use also the ultimate KNX and knxeasy plugin. 
The knxeasy plugin is the core of the problem. After i disable it everything works well.

farmio, 2021-05-12:
Great! 

These NodeRed plugins have their own knx stack and therefore their own connection. Most probably the max concurrent tunneling connections of your interface were used. 

I think we can close this issue.

Blade0001, 2021-05-12:
yes i agree with you

closed 2021-05-12 by farmio