XKNX/xknx #698: Light: Color modes

farmio, 2021-05-18:
- [ ] add support for yxY color mode - DPT242.600 (6byte)
- [ ] move encoding/decoding of RGB and RGBW payloads to DPT module
- [ ] RGBW currently accepts 4-6 bytes payload. The DPT specifies 6bytes. I think we may not accept 4 and 5 bytes for this.

<hr>

- [ ] add RemoteValueColor module using one of these 3 dpts.
- [ ] use a color-mode attribute to select used dpt instead of group addresses for each mode (RGB, RGBW, XY)
- [ ] maybe add auto sense dpt option for color temperature based on payload length 
for RGB, RGBW - XY has the same payload length as RGBW so this would collide

<hr>

- [ ] use a color_temperature-mode attribute to select used dpt instead of group addresses for each mode (CT, TW)
- [ ] add auto sense dpt option for color temperature based on payload length

<hr>

- [ ] maybe add HSV individual address support. I have not found a DPT for this mode - maybe it is a MDT proprietary thing.

crazyfx1, 2021-10-01:
>  use a color_temperature-mode attribute to select used dpt instead of group addresses for each mode (CT, TW)

Is this still something you want to have implemented? I could do that in the next weeks ;)

farmio, 2021-10-01:
@crazyfx1 this is just a collection of ideas I had when I started to change something in the light module. 
If you think this change is useful, go for it 👍. I'm fine with the current implementation too, but a combined address would allow to 
> add auto sense dpt option for color temperature based on payload length

HA already uses a combined address and a mode option.

closed 2022-04-22 by github-actions[bot]