XKNX/xknx #709: Add missing DPT 20.103

cryptedx, 2021-06-06:
Can you please add DPT 20.103 DHW Mode? I haven't found it in documentation.

Thanks in advance!

farmio, 2021-06-06:
In which context do you want to use this? HA?

cryptedx, 2021-06-06:
My Vaillant Heating connected to [ISE Smart](https://www.ise.de/en/products/smart-connect-series/vaillant) has this operating mode for warm water. I want to be able to set it from Home Assistant with expose. "- type: [DHW Mode, 20.103]"

farmio, 2021-06-06:
Can you map your devices functions to the ones of the HA water heater entity? 
https://developers.home-assistant.io/docs/core/entity/water-heater
https://github.com/home-assistant/core/blob/dev/homeassistant/components/water_heater/__init__.py

As a workaround you can use a HA script / input_select etc. to `knx.send` service a generic DPT5 value.
0 = Auto
1 = LegioProtect
2 = Normal
3 = Reduced
4 = Off/FrostProtect 
5..255 = reserved

cryptedx, 2021-06-08:
It doesn't look like that I can map it to HA water heater entity.

But I have done it the other way. Here is my solution:

```yaml
alias: LWWP Heatingzone 1 Operation Mode
description: ''
trigger:
  - platform: state
    entity_id: input_select.lwwp_heizungszone_1_betriebsart
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: input_select.lwwp_heizungszone_1_betriebsart
            state: "Auto"
        sequence:
          - service: knx.send
            data:
              address: 2/3/54
              type: 1byte_unsigned
              payload: 0
      - conditions:
          - condition: state
            entity_id: input_select.lwwp_heizungszone_1_betriebsart
            state: "Comfort"
        sequence:
          - service: knx.send
            data:
              address: 2/3/54
              type: 1byte_unsigned
              payload: 1
      - conditions:
          - condition: state
            entity_id: input_select.lwwp_heizungszone_1_betriebsart
            state: "Standby"
        sequence:
          - service: knx.send
            data:
              address: 2/3/54
              type: 1byte_unsigned
              payload: 2
      - conditions:
          - condition: state
            entity_id: input_select.lwwp_heizungszone_1_betriebsart
            state: "Economy"
        sequence:
          - service: knx.send
            data:
              address: 2/3/54
              type: 1byte_unsigned
              payload: 3
      - conditions:
          - condition: state
            entity_id: input_select.lwwp_heizungszone_1_betriebsart
            state: "Off/BuildingProtection"
        sequence:
          - service: knx.send
            data:
              address: 2/3/54
              type: 1byte_unsigned
              payload: 4
    default: []
mode: single
```

Now I have to synchronise it with the right state from KNX bus in case I change it on the heating control unit itself...

closed 2022-02-16 by github-actions[bot]