XKNX/xknx #710: Could not serialize DPTValue1Ucount value="3.0"

cryptedx, 2021-06-06:
I have to send a DPTValue1Ucount from HA to KNX. I have created this input_number:

```yaml
input_number:
  lwwp_kuehltage_manuelles_kuehlen:
    name: LWWP Kühltage manuelles Kühlen
    min: 0
    max: 255
    step: 1
    mode: box
    unit_of_measurement: "Tage(e)"
    icon: hass:snowflake
```
This is the complete failure:

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/knx/expose.py:102
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 19:51:57 (1 occurrences)
Last logged: 19:51:57

[547317045088] <ConversionError description="Could not serialize DPTValue1Ucount" value="3.0"/>
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/xknx/dpt/dpt_1byte_uint.py", line 43, in to_knx
    knx_value = int(value)
ValueError: invalid literal for int() with base 10: '3.0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1491, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1526, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 728, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/knx/__init__.py", line 291, in reload_service_handler
    await async_setup(hass, config)
  File "/usr/src/homeassistant/homeassistant/components/knx/__init__.py", line 227, in async_setup
    create_knx_exposure(hass, knx_module.xknx, expose_config)
  File "/usr/src/homeassistant/homeassistant/components/knx/expose.py", line 42, in create_knx_exposure
    exposure = KNXExposeSensor(
  File "/usr/src/homeassistant/homeassistant/components/knx/expose.py", line 77, in __init__
    self._init_expose_state()
  File "/usr/src/homeassistant/homeassistant/components/knx/expose.py", line 102, in _init_expose_state
    self.device.sensor_value.value = (
  File "/usr/local/lib/python3.8/site-packages/xknx/remote_value/remote_value.py", line 108, in value
    self.to_knx(value)
  File "/usr/local/lib/python3.8/site-packages/xknx/remote_value/remote_value_sensor.py", line 64, in to_knx
    return DPTArray(self.dpt_class.to_knx(value))
  File "/usr/local/lib/python3.8/site-packages/xknx/dpt/dpt_1byte_uint.py", line 48, in to_knx
    raise ConversionError("Could not serialize %s" % cls.__name__, value=value)
xknx.exceptions.exception.ConversionError: <ConversionError description="Could not serialize DPTValue1Ucount" value="3.0"/>
```

I think that HA is generally sending it as a float. Maybe xknx has to check if it can be converted from float to int in case DPT is DPTValue1Ucount or any similiar DPT which is int only. If it's possible then it should be converted. Then int() should work.

farmio, 2021-06-06:
> knx_value = int(value)

We are already converting to int. It seems like HA passes a string as value of an input_number 🙄

cryptedx, 2021-06-07:
So what do you need from me to support you for debugging?

farmio, 2021-06-07:
It's ok. The thing I was just writing (#711) will help fixing this properly. 
Meanwhile you can use knx.send in a service or template instead of expose.

cryptedx, 2021-06-08:
Ok great. It's not urgent. I will wait for your fix.

Thank you for your time!

closed 2021-06-09 by farmio

cryptedx, 2021-06-13:
It's still not working. Same error.

farmio, 2021-06-13:
Are you using the current dev branch?

cryptedx, 2021-06-13:
Oh no. Sorry. I thought it has been released.