XKNX/xknx #745: [FR] Enable HA Energy Integration

DirkMu, 2021-07-29:
**Description of problem:**

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.18.9 / 2021.8.0b1

**KNX installation:**
I am using a KNX electricity meter (Lingg & Janke: BCU-EZ-KAM-OMNI-OPT1- D-REG-FW) that reads the current vaule from my domestic elec. meter [https://lingg-janke.de/en/knx-smart-metering-products/electricity-meters/](url). The corresponding values (DPT 13.010 and 13.013 beside others) are represented as sensors in HA. 

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
HA 2021.8 provides a new energy integration (either default_config is used or energy is added to configuration.yaml). As per the developers the sensors to be used need to have specific attributes. otherwise they will not appear in the energy configuration: 

- device_class: energy
- state_class: measurement
- last_reset: [timestamp], if never reset 1970-01-01T00:00:00.000000+00:00 

I can add the attributes via customize.yaml however as per the developers those should be provided via integration. Can you provide the attributes via integration, please? 

My meter also allows using a forward counter (DPT 13.013) that can be reset via KNX (DPT 1.017). It would be great if the reset timestamp from the bus could be used for the last_reset

farmio, 2021-07-29:
- we already have the `device_class`.
- `state_class` could imho be added to yaml config of a sensor and maybe even default to `measurement`
- `last_reset` is going to be more tricky. 
Should it only apply to energy device_class?
The reset thing even more. We don't have any historical data so we don't know the last reset time eg. after a restart. Maybe the resettable counter is better solved by a template sensor resetting on an `knx_event` or just setting `last_reset` to current time when the value changes to `0` (Don't think such a stunt would be accepted in HA.core).

@DirkMu would you like to give it a shot adding `state_class` to the integration yourself?

DirkMu, 2021-07-30:
I would try to add it myself. However, I am not able to 🤷‍♂️

I already added state_class and last_reset to my sensors via customize.yaml. This works at the moment. According to Frenck this is not accepted. 

Is last_reset not restored automatically by HA. Currently I use 1970-01-01T00:00:00.000000+00:00 and it is restored after reboot.

farmio, 2021-07-30:
We currently don't restore anything for sensor entities. Current knx sensors have `last_reset = None` (default value for HA SensorEntity class) so what could they restore...
Is it restoring on your installation because of your changes? 

You should be able to use a `power` DPT as input for the integration integration and use this for energy.

Tommatheussen, 2021-07-30:
I'm currently playing around with this as well, I got all of my electrical circuits monitored via KNX, they send both the power and energy objects to the bus. I was able to get the sensors to show up in the Energy dashboard, by adding the `state_class` and `last_reset` as customized attributes, I believe that's what @DirkMu is referring to when 'restored' is mentioned.

As for the `last_reset` attribute, I think it would be fine (at least for now) to make it non-resetable and default to `1970-01-01T00:00:00.000000+00:00`, it can always be reset by deleting the entity and adding it again (I think?)

farmio, 2021-07-30:
For me it's fine, but I don't think this would pass HA review. I had a chat with Frenck, he meant even yaml-configurable `state_class` wasn't a good idea. 

Let's see how this plays out. I guess a lot of requests will come in to have templates be accepted (if they aren't currently, not sure). 
Also Modbus, which seems to be widely used for energy metering, will have to find a solution. Afaik they face the same problems we do - everything but the actual value is user configured.

DirkMu, 2021-07-30:
@Tommatheussen With restore I assumed the last value of `last_reset` is restored automatically after HA was rebooted. If I understand @farmio right no knx values are restore but read from the BUS after HA rebooted.  

@farmio using integration `integration` is something I currently would like to avoid since I reveive the right values from the BUS. Template sensors cannot be used since they don't allow addig attribute `last_reset`. Unless it is added via `customize.yaml`. As you said we're going round in circles. 

wouldn't it be possible to add those values to the knx sensors using `type active_energy_kwh`? As @Tommatheussen said, `last_reset` could be defaulted to `1970-01-01T00:00:00.000000+00:00`. Do you think this would't pass HA review? Do you know, how this is solved by all the other integrations such as shelly etc.?

farmio, 2021-07-30:
> no knx values are restore

not for sensors ;) But it would be a possible route to use HAs restore function for last_reset.

> Do you think this would't pass HA review?

I honestly don't know. 

> Do you know, how this is solved by all the other integrations such as shelly etc.?

I guess with a Shelly you know the model and thus what value type your sensor has and since it has a modern API you probably can also request data like `last_reset` from the device. 

But with our protocol from the late 80s this isn't that easy. An `active_energy` type knx sensor can be a measurement, threshold, historical data, etc. we just don't know unless the user configures it. I'd look for Modbus as it has quite similar restrictions as KNX and is widely used in energy metering (solar inverters, energy meters etc...).

I've looked at the latest ABB actuators, these provide readable timestamps for the last reset (DPT 10.001, DPT 11.001 or DPT 19.001). These could be used in the current sensor implementation/configuration or in a completely new knx device type that generates a sensor in HA useable for energy.

Tommatheussen, 2021-08-02:
FYI, the KNX topic actually came up on the Home Assistant discord channel earlier: https://discord.com/channels/330944238910963714/857346526527094814/871799696161656922

farmio, 2021-08-06:
First step done ☑️  https://github.com/home-assistant/core/pull/53996

closed 2021-09-28 by farmio