XKNX/xknx #765: Expose: send periodically, throttle

farmio, 2021-09-27:
- add an option to send expose values periodically. 
- add an option to suppress too fast updates of exposed values 

Maybe from HA integration

apoc4lyps, 2022-09-24:
Maybe you could consider to send the saved states on startup of Home Assistant to the KNX bus.

farmio, 2022-09-24:
Hi 👋! 
What would be the usecase for this? 
On a HA restart the previous value should already have been sent on its last update before the shutdown.

apoc4lyps, 2022-09-24:
Hi,
I use a binary sensor to lock a motion sensor. After a powerloss the motion sensor does not pick up the right state of the lock GA.
HA should announce the current state of the GA during it startup.

ottonet, 2022-12-14:
Another use case is exposing time and date to the KNX-bus. There are plenty of KNX sensors with displays that require cyclic sending of time and date to ensure the display always shows the right time. (internal clocks of these devices are not very reliable)

ottonet, 2022-12-14:
@farmio Hi ;) 

Good to know. Sadly the HA Expose documentation does not mention this. 

At the same time, there are a few other use cases, where it would make sense. For instance Valve Drives for heating systems, can be set up to monitor the actuating value. If a value is not received in a certain time, it would open to a predefined position. This is to avoid total shutdown of a heating system, in case a thermostat is not functioning properly. 

[image]

I'm actually planning to use a virtual thermostat in HA in the near future, where this thermostat would control a few KNX valves. In case HA crashes or something else is off, and the valve does not receive a new instruction within "X" amount of time, it would move itself to some kind of emergency postion, off or for instance 25%..

farmio, 2022-12-27:
So #1094 solved the main issue here. 

I think for periodical sending and other, more specialized usecases in HA an automation with `knx.send` is better suited than `expose`.
I'm still considering to add "send on startup".

Documentation of expose time is also fixed.

ottonet, 2022-12-28:
"send on startup" is actually quite useful after situations like a power failure. Assuming the KNX system is up before HA, it would ensure that KNX devices always receive the correct value of an object after such a situation, as a read-request after power-on from the device to HA (which is still starting up) would fail in such a case. 

I'm currently in a process of moving my logic and automations in my Gira HomeServer (in use since 2004) to Home Assistant, and I notice I have about 5 different implementations of cyclic sending. In the Gira HomeServer this is setup as an automation (logic building block) as well. 

Obviously it's not too difficult to built such an automation in HA, yet from an end-user / integrator perspective, adding a few lines of YAML in knx_expose.yaml  is much easier / more intuitive, and would cover 99% of the use-cases. 

Example: 
send_on_start: true 
cyclic_sending: true
cycle_time: 3600

farmio, 2022-12-28:
It should probably be `send_on_connection` or something then. As we should also send when HA is running but knx was reconnected - as it would be after a bus reset or something.
I guess this wouldn't work on routing though 🤔

fkzle, 2022-12-28:
I also think that 1 or 2 lines of YAML are a lot easier to configure than generating an automation for every value that should be sent cyclically. 

My usecase is also heating actuators, but different than that of ottonet. 
I am using the controller inside the KNX heating actuator while the temperature sensors are sending their values via MQTT. I need to forward these temperature values through homeassistant to KNX. When the door to a room is closed, the temperature changes so little that no update is sent to the heating acutator. After 1h (this is configurable), the actuator assumes the temperature sensor is gone and goes to some fail-safe mode where the valves are 50% open. 
I really don't like to disable this fail-safe feature, but without periodic sending of the temperatures,  i see no other way. 

The other possible way of implementing this could be: 
if a (exposed) sensor has the "force_update == true" flag set, it is sent to KNX whenever a new value is received.