XKNX/xknx #787: [FR] DPT 1.017 implementation

DirkMu, 2021-10-30:
**Description of problem:**
It appears that Frenk is working on momentary switch which seems perfect for DPT 1.017. 
[https://github.com/home-assistant/core/pull/57642](url)

Can you please add support for DPT 1.017 when the momentary switch is available. 

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.18.11 / 2021.11

**KNX installation:**
./.

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
./.

**Traceback (if applicable):**
./.

farmio, 2021-10-30:
Yey! About time!
I guess we'll implement a more generic way than just sending binary 1 then...

farmio, 2021-11-13:
Coming next release https://github.com/home-assistant/core/pull/59082 - generic as promised 👍

closed 2021-11-13 by farmio

DirkMu, 2021-11-14:
great 👍 thanks