XKNX/xknx #804: Github pages

yugeshk, 2021-11-18:
Hi,

sorry for bothering you guys, but I'm confused as to how you've setup your github pages. For example, you do an `{% include head.html %}` in `docs/_layouts/default.html` but you don't have any `_includes` folder at all?
Am I missing something?

Thanks!

marvin-w, 2021-11-18:
Hi,

It's part of the `just-the-docs` Gem.

https://github.com/pmarsceill/just-the-docs
i.e: https://github.com/pmarsceill/just-the-docs/blob/master/_includes/head.html

yugeshk, 2021-11-18:
So I simply copied your docs source to make a website but github pages doesn't allow me to deploy it.
Only content is changed. You can see the repo here - https://github.com/yugeshk/systems-website
[image]

marvin-w, 2021-11-18:
> So I simply copied your docs source to make a website but github pages doesn't allow me to deploy it.
Only content is changed. You can see the repo here - https://github.com/yugeshk/systems-website

Then now is the right time to actually learn something about what you've copied :).

We don't use github pages, but netlify for deployments, so can't really help there. Also, this is an issue tracker and not a support forum. Please join our discord if you need help.

closed 2021-11-18 by marvin-w