XKNX/xknx #815: No outgoing messages to KNX with Calimero Server as KNX IP router

aschamberger, 2021-12-06:
**Description of problem:**

The problem is that HA can only recieve incoming KNX traffic and no outgoing group communication is possible. This is the same for TP messages routed to IP or messages sent from ETS via IP to the router.
ETS is working fine with the used IP router software: Calimero Server.

HA is running on Home Assistant OS connected via a Calimero Server Addon to KNX.
@see: https://github.com/aschamberger/hassio-addons/blob/master/calimero-server/README.md
My Calimero Server (latest version 2.5) instance is connected via a Weinzierl KNX BAOS Module 838 kBerry (sitting on an Odroid C4).

I have attached a [home-assistant.log](https://github.com/XKNX/xknx/files/7663485/home-assistant.log) where you can see the non working group read for a configured group address on startup. 
Also exposing the time is not working (look for 8/1/*). Same for sending with knx.send service (look for 1/1/8). No messages on the bus. Messages from KNX can be received without problems and state is updated in HA as expected (the 4/2/1 in the config below).

I don't know if it's really a xknx issue or due to the HA integration or something else. As the ETS is working with Calimero Server for all functions I assume the problem is probably on the HA/xknx side.

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: core-2021.11.5 on Home Assistant OS 6.6
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**

```
logger:
  default: warning
  logs:
    homeassistant.components.fritz: error
    fritzconnection: error    ## it seems this one is the problematic
    fritzprofiles: error    
    # For most debugging needs `xnx.log` and one of `xknx.knx` or `xknx.telegram` are a good choice.
    #xknx: debug  # sets the level of all loggers
    xknx.log: debug  # provides general information (connection, etc.)
    #xknx.raw_socket: debug  # logs incoming UDP frames in raw hex format
    xknx.knx: debug  # logs incoming and outgoing KNX/IP frames at socket level
    xknx.telegram: debug  # logs telegrams before they are being processed at device level or sent to an interface
    #xknx.state_updater: debug  # provides information about the state updater

knx:
  individual_address: 1.1.105
  routing:
  expose:
    - type: 'time'
      address: '8/1/0'
    - type: 'date'
      address: '8/1/1'
    - type: 'datetime'
      address: '8/1/2'
  binary_sensor:
    - name: "Haustür offen"
      state_address: "4/2/1"
      sync_state: "expire 10"
#      sync_state: false
      device_class: "door"
      invert: true
```

**Traceback (if applicable):**

farmio, 2021-12-07:
Hi 👋!

This looks very interesting, I did not ever try calimero - maybe we can somehow leverage this as KNX secure proxy for HA 😃

```python
2021-12-06 21:33:54 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="SEARCH_RESPONSE" Reserve="0" TotalLength="78" />
 body="<SearchResponse control_endpoint="<HPAI 192.168.178.79:3671 />" dibs="[
<DIBDeviceInformation 
	knx_medium="KNXMedium.TP1" 
	programming_mode="False" 
	individual_address="1.1.0" 
	installation_number="0" 
	project_number="0" 
	serial_number="00:00:00:00:00:00" 
	multicast_address="224.0.23.12" 
	mac_address="00:1e:06:48:34:91" 
	name="Hass OS KNXnet/IP Server" />,
<DIBSuppSVCFamilies families="[DIBServiceFamily.CORE version: 2, DIBServiceFamily.DEVICE_MANAGEMENT version: 2, DIBServiceFamily.TUNNELING version: 2, DIBServiceFamily.ROUTING version: 2]" />
]" />" />
```
- It seems to also support tunneling. 
Does a tunneling connection work in favour or routing?
- The medium of 1.1 seems to be TP, HA is set to use 1.1.105 with routing. A routing connection should be on an IP line. Maybe HA outgoing telegrams are filtered somewhere.
- Calimero and HA run on the same physical device. Maybe Calimero ignores "outgoing" multicast datagrams on socket level (`IP_MULTICAST_LOOP`). 
Does it work if you run Calimero and HA on different devices?
- This Addon seems to exist for over a year, did it work before?

aschamberger, 2021-12-07:
- Does a tunneling connection work in favour or routing? --> Tunneling also does not work.
- A routing connection should be on an IP line. Maybe HA outgoing telegrams are filtered somewhere. -->  I don't get this. Limited KNX knowledge here ;-)
- Does it work if you run Calimero and HA on different devices? --> Good idea. I'll try this.
- This Addon seems to exist for over a year, did it work before? 
--> Getting to this state did cost me already a lot of time until it worked at least for ETS and I used it only for incoming messages until now on HA. The maintainer of Calimero already fixed some stuff and helped to make me run it with the kBerry and on Docker. With the kberry and ODROID C4 I now have a very compact and also cheap device in a 4 TE case :-)

farmio, 2021-12-07:
> Tunneling also does not work.

Can you post your connection settings and the resulting log when using tunneling? It doesn't use multicast but unicast so we can maybe rule something out.

Does calimero provide any (socket-level) logs? I guess you are running 1 line and the calimero device it attached to this via TP - so the only instance providing some kind of filtering would be calimero. It would be a good idea to see if outgoing HAs telegrams even reach the calimero-container.

aschamberger, 2021-12-07:
I'll generate a log for tunneling tonight. 

What I can say already is that I didn't see anything arriving on the Calimero side. For the working ETS traffic I can see all telegrams in the Calimero logs. Yes, I have one line and the kberry is attached to it. 
Hmm, I already set the log level very high, but I don't remember if it was the highest level already. I'll also try this again.

farmio, 2021-12-07:
What's odd is that the SearchRequests seem to be answered by Calimero whereas the RoutingIndications are ignored 🤷

aschamberger, 2021-12-07:
[_tun_local_home-assistant.log](https://github.com/XKNX/xknx/files/7670249/_tun_local_home-assistant.log)
```
knx:
  individual_address: 1.1.105
  tunneling:
    host: 192.168.178.79
    local_ip: 192.168.178.79
```

[_tun_home-assistant.log](https://github.com/XKNX/xknx/files/7670254/_tun_home-assistant.log)
```
knx:
  individual_address: 1.1.105
  tunneling:
    host: 192.168.178.79
```

Now I'll try to run Calimero from another device. Will post this when I'm ready.

aschamberger, 2021-12-07:
[home-assistant.log](https://github.com/XKNX/xknx/files/7670611/home-assistant.log)

Running Calimero on another device works. Now I remember also that I tested sending messages before with exactly the test device I used now - not running Calimero Server in the container on exactly the same device as HA. I only added later on another kberry to my production HA instance.

Now the question remains where the problem is located that on the same device nothing reaches Calimero ... 
@bmalinowsky maybe you can support us with the problem here.

farmio, 2021-12-08:
In your tunneling tests the `ConnectRequest` frames correctly yield a `ConnectResponse` - so the tunnel is actually set up fine. But then no `TunnelingRequest` is `ACK`ed and xknx closes the connection.
The odd thing about this is that the `ConnectRequest` and the `TunnelingRequest` frames are sent form  and to the same socket - so the communication works at least once. 

I would not see how xknx could be the source of this. We log at socket level - and there are just no answers whatsoever after the `ConnectResponse`. There are also no Link-layer confirmation frames seen (which we don't require, but still).

You tried routing with different devices, does tunneling also work from different devices?

aschamberger, 2021-12-08:
Tunneling does not work with the second device. I'll open a defect at Calimero Server.

```
knx:
  individual_address: 1.1.105
  tunneling:
    host: 192.168.178.72
```
[home-assistant.log](https://github.com/XKNX/xknx/files/7678521/home-assistant.log)

```
knx:
  individual_address: 1.1.105
  tunneling:
    host: 192.168.178.72
    local_ip: 192.168.178.4
```
[home-assistant.log](https://github.com/XKNX/xknx/files/7678538/home-assistant.log)

farmio, 2022-01-19:
@aschamberger xknx 0.19.0 should now be able to tunnel to calimero both via UDP and TCP (this is not yet selectable from HA). Give it a try 👍

aschamberger, 2022-02-05:
Just updated to Home Assistant 2022.2.2 and works :)

closed 2022-02-05 by aschamberger

farmio, 2022-02-05:
Great! Be aware that there is a bug in TCP tunneling connections currently that will be fixed with the next update to xknx #859