XKNX/xknx #820: AttributeError: module 'homeassistant.components.notify' has no attribute 'async_setup_entry'

kalhimeo, 2021-12-14:
**Description of problem:**

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 2021.12.1
- last working xknx / Home-Assistant release (if known): 2021.11.x

**KNX installation:**
- MDT IP Interface (tunelling)

**Full error logs:**

```
Error setting up entry Tunneling @ 192.168.5.2 for notify
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 313, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
AttributeError: module 'homeassistant.components.notify' has no attribute 'async_setup_entry'
```

and (most probably a consequence)

`Exception in discovery_platform_listener when dispatching 'discovery.platform_discovered_load_platform.notify': ({'service': 'load_platform.notify', 'platform': 'knx', 'discovered': [OrderedDict([('name', 'My name 1'), ('address', '10/0/2')]), OrderedDict([('name', 'My name 2'), ('address', '10/0/3')])]},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/helpers/discovery.py", line 119, in discovery_platform_listener await task File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 114, in async_platform_discovered await async_setup_platform(platform, discovery_info=info) File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 88, in async_setup_platform conf_name = p_config.get(CONF_NAME) or discovery_info.get(CONF_NAME) AttributeError: 'NodeListClass' object has no attribute 'get'`

Thanks for your fantastic work on this project

farmio, 2021-12-14:
Hi 👋! 
What does your configuration for the notify platform look like?

kalhimeo, 2021-12-14:
Hi farmio,

Within the configuration.yam (trimmed down for display purpose): 

```
knx:
  notify: !include knx-notify.yaml
```

In knx-notify.yaml :

```
- name: "Z35 Popup Text 1"
  address: "10/0/2"
- name: "Z35 Popup Text 2"
  address: "10/0/3"
- name: "Z35 Popup Text 3"
  address: "10/0/4"
- name: "Fumée Détecteurs Actifs"
  address: "6/1/1"
- name: "Fenêtres et Portes Ouvertes"
  address: "5/2/5"
```

PS: in the error log above I trimmed down the list of addresses too ( OrderedDict([('name', 'My name 2') etc.)

farmio, 2021-12-14:
Aha. Any 
```yaml
notify:
  - platform: knx
``` 
At root level maybe? 

Trimming and changing log messages in a bug report doesn't really make debugging easier, please don't do that.

kalhimeo, 2021-12-14:
I tried your suggestion :

```
notify:
  - name: "Z35 Popup Text 1"
    platform: knx
    address: "10/0/2"
  - name: "Z35 Popup Text 2"
    platform: knx
    address: "10/0/3"
  - name: "Z35 Popup Text 3"
    platform: knx
    address: "10/0/4"
  - name: "Fumée Détecteurs Actifs"
    platform: knx
    address: "6/1/1"
  - name: "Fenêtres et Portes Ouvertes"
    platform: knx
    address: "5/2/5"
```

Gives me this error in HA log :

```
Logger: homeassistant.components.notify
Source: components/notify/legacy.py:75
Integration: Notifications (documentation, issues)
First occurred: 13:29:10 (5 occurrences)
Last logged: 13:29:10

Failed to initialize notification service knx
```

Then I set everything back with the notify declared at knx level, here are the 2 full logs : 

```
Logger: homeassistant.config_entries
Source: config_entries.py:313
First occurred: 13:33:22 (1 occurrences)
Last logged: 13:33:22

Error setting up entry Tunneling @ 192.168.5.2 for notify
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 313, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
AttributeError: module 'homeassistant.components.notify' has no attribute 'async_setup_entry'
```

and

```
Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 13:33:22 (1 occurrences)
Last logged: 13:33:22

Exception in discovery_platform_listener when dispatching 'discovery.platform_discovered_load_platform.notify': ({'service': 'load_platform.notify', 'platform': 'knx', 'discovered': [OrderedDict([('name', 'Z35 Popup Text 1'), ('address', '10/0/2')]), OrderedDict([('name', 'Z35 Popup Text 2'), ('address', '10/0/3')]), OrderedDict([('name', 'Z35 Popup Text 3'), ('address', '10/0/4')]), OrderedDict([('name', 'Fumée Détecteurs Actifs'), ('address', '6/1/1')]), OrderedDict([('name', 'Fenêtres et Portes Ouvertes'), ('address', '5/2/5')])]},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/helpers/discovery.py", line 119, in discovery_platform_listener await task File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 114, in async_platform_discovered await async_setup_platform(platform, discovery_info=info) File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 88, in async_setup_platform conf_name = p_config.get(CONF_NAME) or discovery_info.get(CONF_NAME) AttributeError: 'NodeListClass' object has no attribute 'get'
```

closed 2021-12-15 by marvin-w