XKNX/xknx #842: Light entity reports incomplete supported color modes

crazyfx1, 2022-01-06:
**Description of problem:**
I have a light which supports color and color_temperature. Both addresses are set.

But the light entity only reports one of the modes as supported color modes. The entity should report all supported modes.

If both color and color_temp are set, the following state is reported:
```yaml
supported_color_modes:
  - rgb
friendly_name: Schlafzimmer Licht
supported_features: 0
```

If I remove the color part, the color_temp is returned:
```yaml
min_mireds: 166
max_mireds: 370
supported_color_modes:
  - color_temp
friendly_name: Schlafzimmer Licht
supported_features: 0
```

The hue integration for example reports both types:
```yaml
min_mireds: 153
max_mireds: 500
supported_color_modes:
  - color_temp
  - xy
```

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 2021.12.5

**KNX installation:**
Not relevant

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
```yaml
    - name: "Schlafzimmer Licht"
      address: '2/1/17'
      state_address: '2/2/17'
      brightness_address: '2/3/17'
      brightness_state_address: '2/4/17'
      color_temperature_address: '3/1/17'
      color_temperature_state_address: '3/2/17'
      color_temperature_mode: absolute
      color_address: '3/3/17'
      color_state_address: '3/4/17' 
```

**Traceback (if applicable):**

farmio, 2022-01-06:
Which actuator are you using for this?
HA would require to be able to switch between color_modes somehow. How does your actuator handle this?

crazyfx1, 2022-01-06:
It's the ISE Hue KNX gateway again ;)

I don't know how it handles the switch internally, what exactly do you want to know? 
The color and color_temperature are separate GA and depending on which GA I'm sending the mode is changed.

farmio, 2022-01-06:
How does it work from a users point of view?
- when you have the light set to a color and change color temperature
  - does it change to white then? 
  - is the brightness kept?
- does it update the color_temperature GA when changing rgb-color? What would a GroupValueRead yield?
- does it update the rgb-color GA when changing color_temperature? What would a GroupValueRead yield?
- what does it do on a simple (DPT 1) turn_on? Is that configurable?
- is there a way to switch between color_temperature mode and color mode without changing one of the values (eg. from an App)?

crazyfx1, 2022-01-07:
> How does it work from a users point of view?
> 
> * when you have the light set to a color and change color temperature
>   
>   * does it change to white then? **yes, color temp and color are two different pallettes**
>   * is the brightness kept? **yes**
> * does it update the color_temperature GA when changing rgb-color? What would a GroupValueRead yield? **not actively. if you read it with GroupValueRead the device converted the rgb to color_temp (not very accurately since its not really possible).**
> * does it update the rgb-color GA when changing color_temperature? What would a GroupValueRead yield? **not actively. if you read it with GroupValueRead the device converted the color_temp to rgb (quite accurate)**
> * what does it do on a simple (DPT 1) turn_on? Is that configurable? **yes, the value for turn_on is configured in the hue app)**
> * is there a way to switch between color_temperature mode and color mode without changing one of the values (eg. from an App)?
**yes, you can switch in the hue app and also HA supports switching the modes. In HA I think it's only in the frontend, nothing happens if you switch between the tabs, only when you select a color.**
[image]
[image]

I've also checked the code from the HA hue integration. They also try to guess the current color_mode, I think it's for the frontend to show the correct mode when opening the popup. Other than that I don't see it used anywhere.

farmio, 2022-01-07:
Just out of interest: why are you using the extra step over KNX when HA could leverage Hues API directly (I have no experience with Hue whatsoever)?

The state updater could cause problems with a set color_mode. Other than that I guess switching mode based on last received rgb/ct GA could be a viable option. 
Maybe these should only init, not refresh state - then the last color value would also be stored instead of being converted. (manual `sync_state: init` would be fine imho since this is such a niche problem). 

On turn on - does it send a rgb / ct state value actively based on internal mode?

crazyfx1, 2022-01-07:
I'm using the adaptive lighting component in HA which is basically HCL and sets the color temperature based on daytime. When I turn on the light at the wall switch, the ISE Hue Gateway reports the "on" state to KNX/HA and the color temperature could be set without delay.
The old HA hue integration used polling, so there was a delay when using that. But recently the hue integration was updated to the v2 push API and I would need to try out if that works better now. Then I could migrate to the hue integration.

The other question I need to check tomorrow.

farmio, 2022-01-07:
I have one more:
In the Hue integration: when you change mode, does the bulb change immediately or just after you pick a color?

Wouldn't this adaptive lighting thing mess with colors then when it sends a new value? Eg. when a light is set to an rgb color and a new CT is broadcasted?

closed 2022-04-22 by github-actions[bot]