XKNX/xknx #877: Request info in tunneling devices

sergioc99, 2022-02-21:
Hi!! 

I am using the library to do a job, I am doing send_telegram by tunneling to a KNX device to set a status, for example to turn on lights, but I am a bit lost when it comes to get the status of a device (to know if they are on or off). Any help?

Thank you in advance! :)

   ```
await tunnel.connect()

    await tunnel.send_telegram(
            Telegram(
                destination_address=GroupAddress(luz),
                payload=GroupValueWrite(DPTBinary(switch)),
            )
        )
    await tunnel.disconnect()
```

That's what I use to send a telegram.

farmio, 2022-02-21:
Hi 👋!
Have a look at this example https://github.com/XKNX/xknx/blob/main/examples/example_light_state.py
It uses a Light device instead of sending the telegram directly via a Tunnel class.

sergioc99, 2022-02-22:
Hiii! Thanks for your faster answers, I don't know why but I can only connect by tunnel so I have tried low-level approach and I get this error: logger.warning(
                "Error: KNX bus did not respond in time (2 secs) to GroupValueRead request for: 1/1/1"

Any idea what is happening? :( 

Thank you again!

farmio, 2022-02-22:
This (and most other GroupAddress related things in xknx) works over the TelegramQueue. Using only a Tunnel object directly doesn't read from it or put its received telegrams.
You really shouldn't use a Tunnel object at all - xknx is not meant to be run that way. 

Start an XKNX object properly and use a ConnectionConfig class if automatic connection doesn't work for you.
Maybe see this example: https://github.com/XKNX/xknx/blob/main/examples/example_manual_connection.py

sergioc99, 2022-02-25:
I had done some tests and I always get a null or None response, I am trying this rn.

```
"""Connect to specific tunnelling server, time a request for a group address value."""
    connection_config = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING,
        gateway_ip="192.168.7.206",
        gateway_port=3671,
        local_ip="192.168.7.12",
        local_port=0,
        route_back=True,
    )
    xknx = XKNX(connection_config=connection_config)

    light = Light(xknx, name="TestLight", group_address_switch="1/1/1")
    await light.set_on()
    await asyncio.sleep(2)
    await light.set_off()
    print(light)
    return
```

and this is the answer

<Light name="TestLight" switch=<1/1/1, None, [], None /> />

farmio, 2022-02-25:
You need to call `await xknx.start()` before, or run in an async context manager (`async with xknx:`). 
Have a closer look to the examples or the documentation at xknx.io

closed 2022-03-09 by farmio