XKNX/xknx #878: Request: support DPT1.007 (1bit) setpoint shift mode for climate component

Olum-hack, 2022-02-24:
I use the MDT Heating Actuators with the 1Bit setpoint shift mode. 
That's easy to handle because all the values are inside of the controller. The other operating panels or buttons don't need any additional information about the setpoint (step or range)

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.19.2 / 2022.2.9
- last working xknx / Home-Assistant release (if known): never

**KNX installation:**
I use this actor from MDT as heating actuators

[https://www.mdt.de/EN_Heating_Actuators.html](https://www.mdt.de/EN_Heating_Actuators.html)

I few years ago there was the same discussion:
https://github.com/XKNX/xknx/issues/420

marvin-w, 2022-02-24:
I actually use the same heating actuator. You can simply change the setpoint shift mode in your parametrization and things will start to work.

farmio, 2022-02-24:
You can use this in HA by configuring it as 2 Button entities. 
For a climate entity everything discussed in #420 still applies.

closed 2022-04-07 by marvin-w