XKNX/xknx #880: Cover device_class not applied

DirkMu, 2022-03-02:
**Description of problem:**
the device_class maintained for knx covers are not applied When shutter is maintained blind will be shown in HA.
[image]

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.19.2
- last working xknx / Home-Assistant release (if known): don't know

**KNX installation:**

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
```- name: bad
  move_long_address: '3/2/111'
  move_short_address: '3/2/113'
  position_address: '3/2/114'
  position_state_address: '3/2/115'
  travelling_time_down: 17
  travelling_time_up: 17
  invert_position: false
  device_class: shutter
```

**Traceback (if applicable):**

farmio, 2022-03-02:
Hi! Which HA core version are you using? I recall there was a hiccup when the device_class definitions were changed to Enum - but this should already be sorted out.

DirkMu, 2022-03-03:
Hi. I am using the latest. 2022.3.0. I never saw this before because in customize.yaml I specified friendly name, device_class etc. Currently I am cleaning up my customize.yaml and hence I added device_class to the KNX cover.

farmio, 2022-03-03:
I have just tested this and it works fine on my system 🤷. Check if you didn't accidentally overwrite the device_class from eg. a customize entry/glob.

DirkMu, 2022-03-03:
I disabled customize.yaml, customize_glob.yaml and customize_domain and it did not work. 

In configuration.yaml I am using 

```
knx:
  binary_sensor: !include knx_binary_sensors.yaml
  button: !include knx_buttons.yaml
  climate: !include knx_climates.yaml
  cover: !include knx_covers.yaml
```

and in knx_covers.yaml I am using
```
- name: bad
  move_long_address: '3/2/111'
  move_short_address: '3/2/113'
  position_address: '3/2/114'
  position_state_address: '3/2/115'
  travelling_time_down: 17
  travelling_time_up: 17
  invert_position: false
  device_class: shutter
```

The device_class of the cover is always blind. No matter what I do. 🤷‍♂️

DirkMu, 2022-03-03:
What's the difference between device_class and original_device_class?

```
{
      "area_id": "badezimmer",
      "capabilities": {},
      "config_entry_id": "7b9c275b508cb9d991f11d84ced1e647",
      "device_class": "blind",
      "device_id": null,
      "disabled_by": null,
      "entity_category": null,
      "entity_id": "cover.bad",
      "icon": null,
      "id": "4eb87155aa556806086b188cfd4a6194",
      "name": "Bad",
      "options": {},
      "original_device_class": "shutter",
      "original_icon": null,
      "original_name": "bad",
      "platform": "knx",
      "supported_features": 127,
      "unique_id": "3/2/111_3/2/114",
      "unit_of_measurement": null
  },
```

farmio, 2022-03-03:
`original_device_class` is the one coming from the integrations whereas `device_class` represents the value a user has customized it to.
Same for `original_name` and `name` etc. I guess so it is reversible, but don't know how exactly that works.

DirkMu, 2022-03-04:
Do you know why the show as field is missing for covers? This is where I control the device_class. When I initialize in core.entity_registry the device_class from knx is applied. 

[image]

[image]

farmio, 2022-03-04:
No idea, sorry. I think you will have better chances to get an answer to this on HA discord or forum. 
Closing this as it is not xknx related.