XKNX/xknx #882: Request: knx2mqtt addon for Home Assistant

CharlieBailly, 2022-03-08:
Hi all,

I am using KNX in Home Assistant with a lot of devices and it is working pretty well, so thanks for the job done here.
However I am also using Zigbee, and first I was using ZHA but I switched to zigbee2mqtt because it is decoupled from Home Assistant.
So I was wondering if you ever considered creating a KNX addon (on top of this `xknx` library) equivalent to the zigbee2mqtt one, because I think it could be great to be able to decouple KNX from Home Assistant.

farmio, 2022-03-08:
Hi! 
I never considered that. 
I guess it would require its own configuration parser and mqtt logic. That's a whole separate project.

farmio, 2022-03-09:
For Home Assistant I don't see any point in adding an abstraction layer by using an addon when we can interact directly with xknx by using the KNX integration. 
For usage apart from HA this may be a viable project idea, but is out of scope for this repository - so I'm closing this feature request.
If you (or anyone else) decide to start working on this feel free to join our discord server so we can talk about it.

closed 2022-03-09 by farmio