XKNX/xknx #891: Problem with RGBW Lights in HA

zoesn89, 2022-03-24:
Hey guys,

I have a problem with controlling my RGBW Strips in my house. I configured four RGBW lights (3 via DALI, 1 via MDT AKD) in the ETS and imported the addresses in my knx.yaml. In the lovelace UI, I can turn on/off the lights, but when I'm changing the color, the brightness, hue or white xknx sends data to the wrong addresses. I analysed the group monitor in ETS and it seems, that the addresses change randomly. E.g. when I'm changing the color from my light "LED Strip Bett", xknx addresses the light "WZ Wand" or "WZ Insel" and vice versa. I've no clue, why this is happening, but it seems, that xknx has some problems, when there are more than one RGBW light.

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: **Home Assistant 2022.3.5**

**KNX installation:**
- Manufacturer and model of relevant actors: **MDT AKD-0424R.02 & MDT SCN-DALI32.03**
- relevant group address parameters: **DPT: 251.600, Flags: Communication, Write**
- excerpt of bus monitor output:
**info: all commands were meant for the light "LED Strip Bett"!**
```
#	Zeit	Dienst	Flags 	Prio	Quelladresse	Quellname	Zieladresse	Zielname	Schnittstelle	Hop Count	Typ	DPT	Info
16	24.03.2022 11:10:03,341	vom Bus		Niedrig	1.0.2	Tunnel 2	2/1/121	LED Strips WZ Wand Farbszenenobjekt		6	GroupValue_Write		00 FF 40 00 00 0F | Log Number=0, Alarm Priority=-1, Application Area=64, Error Class=0, Reserved4=, ErrorCode_Sup=False, AlarmText_Sup=False, TimeStamp_Sup=False, Ack_Sup=False, Reserved9=, Locked=True, AlarmUnAck=True, InAlarm=True

#	Zeit	Dienst	Flags 	Prio	Quelladresse	Quellname	Zieladresse	Zielname	Schnittstelle	Hop Count	Typ	DPT	Info
97	24.03.2022 16:35:09,596	vom Bus		Niedrig	1.0.2	Tunnel 2	2/2/120	LED Strip Bett Schalten		6	GroupValue_Write		$01 | Ein
98	24.03.2022 16:35:09,778	vom Bus		Niedrig	1.1.86	AKD-0424R.02 LED Controller 4 Kanäle/RGBW, REG	2/2/129	LED Strip Bett Status Ein-Aus		5	GroupValue_Write		$01 | Aktiv
99	24.03.2022 16:35:12,135	vom Bus		Niedrig	1.1.86	AKD-0424R.02 LED Controller 4 Kanäle/RGBW, REG	2/2/130	LED Strip Bett RGBW Status Dimmwert		5	GroupValue_Write		00 00 FF 00 00 0F | #0000FF W:0%
100	24.03.2022 16:35:12,241	vom Bus		Niedrig	1.1.86	AKD-0424R.02 LED Controller 4 Kanäle/RGBW, REG	2/2/131	LED Strip Bett HSV Status Dimmwert		5	GroupValue_Write		AA FF FF | AAFFFF
101	24.03.2022 16:35:12,364	vom Bus		Niedrig	1.1.86	AKD-0424R.02 LED Controller 4 Kanäle/RGBW, REG	2/2/134	LED Strip Bett Helligkeit (V) Status Dimmwert		5	GroupValue_Write		$FF | 100%
102	24.03.2022 16:35:12,826	vom Bus		Niedrig	1.0.2	Tunnel 2	2/2/77	LED Strips Bad Dimmen absolut		6	GroupValue_Write		$5E | 37%
103	24.03.2022 16:35:13,034	vom Bus		Niedrig	1.1.3	DaliControl 32	2/2/79	LED Strips Bad Status Dimmwert		5	GroupValue_Write		$5E | 37%
104	24.03.2022 16:35:13,088	vom Bus		Niedrig	15.15.15	-	2/2/79	LED Strips Bad Status Dimmwert		6	GroupValue_Write		$5E | 37%
105	24.03.2022 16:35:13,088	vom Bus		Niedrig	1.1.3	DaliControl 32	2/2/78	LED Strips Bad Status An-Aus		5	GroupValue_Write		$01 | Ein

```

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
```
knx:
  light:
    - name: "WZ Wand"
      address: "2/1/100"
      state_address: "2/1/103"
      brightness_address: "2/1/102"
      brightness_state_address: "2/1/104"
      rgbw_address: "2/1/106"
      rgbw_state_address: "2/1/107"
    - name: "WZ Insel"
      address: "2/1/130"
      state_address: "2/1/133"
      brightness_address: "2/1/132"
      brightness_state_address: "2/1/134"
      rgbw_address: "2/1/136"
      rgbw_state_address: "2/1/137"
    - name: "Bad Strips"
      address: "2/2/75"
      state_address: "2/2/78"
      brightness_address: "2/2/77"
      brightness_state_address: "2/2/79"
      rgbw_address: "2/1/81"
      rgbw_state_address: "2/1/82"
    - name: "LED Strip Bett"
      address: "2/2/120"
      state_address: "2/2/129"
      brightness_address: "2/2/77"
      brightness_state_address: "2/2/125"
      rgbw_address: "2/1/121"
      rgbw_state_address: "2/1/130"
```

Thanks in advance!
Felix

farmio, 2022-03-24:
Hi 👋! 

This behaviour is expected since you seem to have happily mixed the configured GAs for your channels 🤷

> 2/1/121	LED Strips WZ Wand Farbszenenobjekt
> 2/2/77	LED Strips Bad Dimmen absolut
```yaml
    - name: "LED Strip Bett"
      address: "2/2/120"
      state_address: "2/2/129"
      brightness_address: "2/2/77"
      brightness_state_address: "2/2/125"
      rgbw_address: "2/1/121"
      rgbw_state_address: "2/1/130"
```

zoesn89, 2022-03-24:
> Hi 👋!
> 
> This behaviour is expected since you seem to have happily mixed the configured GAs for your channels 🤷
> 
> > 2/2/120	LED Strip Bett Schalten
> > 2/1/121	LED Strips WZ Wand Farbszenenobjekt
> > 2/2/77	LED Strips Bad Dimmen absolut
> 
> ```yaml
>     - name: "LED Strip Bett"
>       address: "2/2/120"
>       state_address: "2/2/129"
>       brightness_address: "2/2/77"
>       brightness_state_address: "2/2/125"
>       rgbw_address: "2/1/121"
>       rgbw_state_address: "2/1/130"
> ```

Damn, you are right, of course! I mixed up two versions of my knx.yaml...
Luckily, I got my new glasses today :D 

Thanks for your help! ;)