XKNX/xknx #912: Special characters not rendered correctly since HA 2022.4

tfriedlx, 2022-04-11:
**Description of problem:**

- [ x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 2022.4.1
- last working xknx / Home-Assistant release (if known): 2022.3.x

I'm using the notify feature to send info messages to my KNX devices - specifically to a MDT Glass Panel. Some of these messages contain special characters (ö, but also signs like °). Since the 2022.4 release those characters are only rendered as ?. Nothing changed on the KNX side since the 2022.4 update so I assume the updates caused the issue.

farmio, 2022-04-11:
Hi 👋! 
What exact DPT is your Knx device expecting?
If it was `16.000` that would be ASCII - which is used by the notification entity. According to the specs (at least how I read it, and how ETS implement it) there are no Umlauts and a very small set of special characters supported. 
We previously had implemented this kind of wrong as we parsed it to an 8-bit value instead of 7. 
This was changed when we added the `latin_1` DPT 16.001 sensor type which supports Umlauts according to specifications. This new sensor type is not yet available to notification service, but can be used from `knx.send` in HA.

tfriedlx, 2022-04-11:
Hi,

Thanks for the quick response! It was 16.000. I switched now to knx.send with latin_1 and it works again. Thanks!

marvin-w, 2022-04-11:
Nice!

marvin-w, 2022-04-11:
Leaving this open until the notify type is available in HA.

closed 2022-04-23 by farmio