XKNX/xknx #928: How to use xknx in Windows or in a VM

18611429192, 2022-04-25:
I want to use xknx in Windows10 with Python3.9.6, but I see this [issue](https://github.com/XKNX/xknx/issues/215#issue-462291584). The same question is mentioned in this question, but I can't see the [link](https://github.com/farmio/xknx/tree/docker-support). 

**Now I can't access the KNX/IP gateway in Windows 10 with Ubuntu 18.04 installed in Virtual Box.**

My Ubuntu with a Host-Only network and ip is 192.168.56.112.

My Windows10 ip is 192.168.56.1, and the KNX/IP gateway in port 3671.

farmio, 2022-04-25:
You can use it from Windows directly if you like. From VM should work just as well. 
The issue you linked only applies when you are not on the same subnet as the Gateway / there is NAT involved. You can use route_back from a ConnectionConfig - environment variables are not used anymore. https://github.com/XKNX/xknx/blob/bb80a063af41919e425710df50788ac5f0d6eed0/examples/example_manual_connection.py#L21

18611429192, 2022-04-26:
Thank you for your reply. 

I tested it in the actual situation yesterday. I used the gateway_scanner.py by `python example_gatewayscanner.py` in Windows10 with python3.9.6, but I didn't find the corresponding results.

My ip is 192.168.1.100

The actual connection is shown in the figure. IPR is connected to the switch. I use network cable to connect the switch.

[image]

At the same time, I can use ETS6(demo) to detect these two IPR.
[image]

farmio, 2022-04-26:
Oh right, I see. Remove / comment out this line: https://github.com/XKNX/xknx/blob/bb80a063af41919e425710df50788ac5f0d6eed0/examples/example_gatewayscanner.py#L14

Seems it remained from a test of a recently added feature (which I'm just refactoring again so it will be corrected soon).

18611429192, 2022-04-26:
Thank you. I will try it later.

If it works, I will close the issue.

farmio, 2022-04-26:
Great! 
Please use copy&paste instead of photos of your monitor then 🙏