XKNX/xknx #930: How to monitor telegram from knx IP Router for long time

18611429192, 2022-04-26:
I want to monitor telegram from knx IP Router for long time, so I use `python .\example_telegram_monitor.py` with a setting like this.
 ``` 
async def monitor(address_filters):
    xknx = XKNX(daemon_mode=False)
    xknx.telegram_queue.register_telegram_received_cb(
        telegram_received_cb, address_filters
    )
    connection_config = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING,
        gateway_ip="192.168.1.101",
        gateway_port=3671)
    xknx.connection_config = connection_config

    await xknx.start()
    await xknx.stop() 
```

The result is direct close, it can't monitor for  long time and it can't also receive a telegram from KNX IP Router.

If I Remove / comment `await xknx.stop() `, it will report a error like this.
```
_GatheringFuture exception was never retrieved
future: <_GatheringFuture finished exception=CancelledError()>
Traceback (most recent call last):
  File "C:\Users\liuli74\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\pyenv-win\versions\3.9.6\lib\asyncio\queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

asyncio.exceptions.CancelledError
Could not close loop, reason: There is no current event loop in thread 'MainThread'.
Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x000001F3BC7311F0>
  File "C:\Users\liuli74\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\pyenv-win\versions\3.9.6\lib\asyncio\proactor_events.py", line 116, in __del__
    self.close()
  File "C:\Users\liuli74\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\pyenv-win\versions\3.9.6\lib\asyncio\proactor_events.py", line 108, in close
    self._loop.call_soon(self._call_connection_lost, None)
kages\Python37\site-packages\pyenv-win\versions\3.9.6\lib\asyncio\base_events.py", line 746, in call_soon
    self._check_closed()
kages\Python37\site-packages\pyenv-win\versions\3.9.6\lib\asyncio\base_events.py", line 510, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
```

farmio, 2022-04-26:
Use `daemon_mode=True` see https://xknx.io/xknx.html

18611429192, 2022-04-26:
I use the `aemon_mode=True`, but xknx notice me 
```
Windows does not support signals
Press Ctrl+C to stop
```

I don't know if xknx can continue to work.

farmio, 2022-04-26:
On Windows ctrl+c Signal just doesn't work. (At least it didn't when this was written). You can just use a different method to keep your program running, but I don't know Windows well so can't really help you there. 
It's a generic Python question, not really xknx related so you'll probably find something in SO or something.

18611429192, 2022-04-26:
If possible I want to monitor telegram forever. I can use     while true to run xknx. So how can I to  modify the code  ? any suggestion?

farmio, 2022-04-26:
You can just wait for an event that never happens. 
```py
await asyncio.Event().wait()
```
Not really sure what the best option is though.

18611429192, 2022-04-27:
Thank you for your advice. I will try it later.

closed 2022-04-27 by 18611429192