XKNX/xknx #983: Minimum hardware for KNX

Maxhera, 2022-06-26:
Hi,
i'm new here and on xknx library, KNX ...
XKNX library and Python is great for me.
I want to know: 
1. what is minimum hardware and software to program KNX devices for home automation?
(Laptop, Python 3..., and...)
2. Are starter kits to get you started on ?

Thanks in advance,
Maxhera
musa_maxhera@web.de

farmio, 2022-06-26:
Hi 👋!
I'm afraid there aren't any starter kits per se (at least what I know of). 
You would need a minimal KNX installation - that consists of a power supply, an IP Interface (or Router) and some device you want to control (like a switch actuator or some sensor).
Then you would also need to get the ETS software to provision these devices. Xknx can't really program devices, but only control / read (well technically it can, but I'd strongly advise against it if you don't 100% know what you are doing).

Maxhera, 2022-06-27:
thank you for your quick reply.
I want to control without ETS software.
As Hadrware platfor I want to use a Raspberry pi 4.
Read digital and analog inputs
Write digital and analog outputs.
Is that possible?

Thank you very much for your help

Maxhera, 2022-06-27:
Hi,
I don't want to use ETS Software(is too expensive, and ....)
My configuration is:
 
STV-0640.02 KNX bus voltage supply 640mA
SCN-IP000.03 IP Interface with Secure
SBS/U6.0.11 HVAC slave device, 6-fold BE
AKU-1616.03 universal actuator 16-fold, 16A, 230VAC

I want to use for example Taster to turn ON/OFF an Actor...

Thank you very much for your help

farmio, 2022-06-28:
You can use ETS Demo for up to 5 devices for free. 
You can't program KNX devices with xknx.
You don't need xknx (nor ETS after initial provision) to toggle a KNX actuator from a KNX swtich.

Maxhera, 2022-06-28:
Hi,
thank you for your time.

if i don't like to use ETS is it possible to 
Read -Inputs and Write- Outputs ?

"""Example for switching a light on and off."""
import asyncio

from xknx import XKNX
from xknx.devices import Light

async def main():
    """Connect to KNX/IP bus, slowly dimm on light, set it off again afterwards."""
    xknx = XKNX()
    await xknx.start()

    light = Light(
        xknx,
        name="TestLight2",
        group_address_switch="1/0/12",
        group_address_brightness="1/0/14",
    )

    for i in [0, 31, 63, 95, 127, 159, 191, 223, 255]:
        await light.set_brightness(i)
        await asyncio.sleep(1)
    await light.set_off()
    await xknx.stop()
asyncio.run(main())

This example ist from:
[xknx-main.zip](https://github.com/XKNX/xknx/files/9002329/xknx-main.zip)

Thanks and have a nice day.

farmio, 2022-06-28:
No. You need to upload the devices Application and link the group objects to group addresses via ETS.

closed 2022-07-01 by farmio