XKNX/xknx #1033: Climate temperature change issues

DirkMu, 2022-09-22:
**Description of problem:**
I don't know if this is specific to my device but I thought it might be worth reporting it here. 

When I have activated a preset mode other than Comfort, such a standby/away or night and I  change the target temperature from home assistant by e.g. 0,5 degrees the value changes by 1,5 degress. I assume this is due to the setting 'target value reduction' set in the knx application (see image 1) which is relative to the value of the value set for comfort. 

When the target temperature is changed from the thermostat directly (preset mode is active) everythings works correctly. 

I don't know if this is specific to the Hager WYT62x or somethings that is valid for other manufactures, too. Further if this can be handled by xknx. As you can see in image 3 19,5 degrees was received from tunnel 2. The thermostat resonded with 18,5 degrees. Any idea how the device itself handles this? 

When preset mode comfort is active everything works fine. 

- [ ] using xknx standalone
- [X] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: 1.0.2
- last working xknx / Home-Assistant release (if known):

**KNX installation:**
Hager WYT62x thermostat

[image]
[image]
[image]

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
```
- name: bad
  temperature_address: '4/2/65'
  target_temperature_address: '4/2/63'
  target_temperature_state_address: '4/2/64'
  operation_mode_state_address: '4/2/62'
  operation_mode_address: '4/2/61'
  temperature_step: 0.5
  command_value_state_address: '4/2/66'
  active_state_address: '4/2/68'
  min_temp: 18
  max_temp: 25
```

**Diagnostic data of the config entry (only when Home Assistant is used)**

**Traceback (if applicable):**
Set a preset mode other than comfort (or whatever the default is) and change the target temperature

farmio, 2022-10-10:
Hi 👋! 
Sorry for the late answer. 

Unfortunately there doesn't seem to be consensus about how target temperatures are set in Knx. Every manufacturer seems to implement their own ideas for that 😭

Is there maybe a way to use setpoint shift group object or individual independent temperatures instead of deviating preset temperatures from comfort?

Otherwise, is there a read flag set by default for the comfort target temperature GO (4/2/63)? If not, does it respond correctly if you set the flag?

DirkMu, 2022-10-15:
Hi.
Sorry for the late response. Unfortunately, there is no setpoint shift GO. The read flag is set be default. I assume there is no way change that behavior.

farmio, 2022-10-15:
Hm... as a workaround you could configure `target_temperature_state_address` to the comfort address (same as `target_temperature_address`) and have a Sensor displaying the current mode setpoint ('4/2/64').

Otherwise it would require an additional attribute of the Climate device like "current_mode_target_temperature" or something like this that overrides https://github.com/home-assistant/core/blob/dev/homeassistant/components/knx/climate.py#L156-L159

DirkMu, 2022-10-15:
To be honest, I think we cannot workaround this outside the thermostat. Since there is no GO, I assume, in order to adapt the target temp of mode e.g. 'night' you have to change to comfort, adapt the value, and go back to night.

farmio, 2022-10-15:
My workaround proposal would create a "Comfort temperature climate entity". It would always show comfort temp regardless of current preset.

DirkMu, 2022-10-17:
I am not sure if I get you correctly. Will this allow me to change e.g. the "night target temperature"? This is what I am attempting to achieve. 

I can change the target temp while preset mode comfort is active. However I cannot change the target temp of night while night is active. Using the physical thermostat this is possible. The physical thermostat allows me to overwrite the value (-2) defined in ETS. It is stored in the thermostat. E.g. comfort is 20 and night by default is 18. When I change the target temp while night is active the temperature of the preset mode night is changed. Comfort remains unchanged. As a result the shift differs from the value defined in ETS. 

Changing the target temp in HA while preset mode night is active will lead to a change of the comfort value. E.g. Comfort is 20, night is 18. Night is active and I decrease the temp by -0.5. HA thus sends 17.5 and the device responds with 15.5, because 17.5 is assumed to be the new comfort value which is decreased by 2.

farmio, 2022-10-17:
No. It seems your controller just doesn't support changing individual mode (preset) temperatures from the bus.

DirkMu, 2022-10-18:
Seems so. Thanks for helping.

closed 2022-10-18 by DirkMu