XKNX/xknx #1034: System telegrams are not properly processed

knxjoerg, 2022-09-23:
I am using xknx in Py3.10 on Win10 with a Weinzierl 731 KNX IP Interface. 

Problem 1: 
Using xknx as a simple bus-monitor system telegrams like T_Connect, T_Disconnect, DeviceDescriptor_Read, DeviceDescriptor_Response, PropertyValue_Read, PropertyValue_Response are not properly passed to the telegram_received_cb callback function. 

Problem 2:
A T_Connect telegram never appears on the KNX bus. I get an "Unhandled management telegram" warning instead. I did not try any further system telegrams, because without connection, they dont make sense. 

Example program:
`
import asyncio
from xknx import XKNX
from xknx.telegram import Telegram
from xknx.telegram.address import IndividualAddress
from xknx.telegram.tpci import TConnect

async def telegram_received_cb(telegram: Telegram):
    print("Telegram received: {0}".format(telegram))

async def main():
    xknx = XKNX(own_address="0.4.63", telegram_received_cb=telegram_received_cb) #, daemon_mode=True)
    await xknx.start()
    telegram = Telegram(source_address="0.4.63", destination_address=IndividualAddress("0.4.6"), tpci=TConnect())
    await xknx.telegrams.put(telegram)
    await asyncio.sleep(60)
    await xknx.stop()

asyncio.run(main())
`

Many thanks, 
Joerg

farmio, 2022-09-23:
Hi 👋! 
Currently only incoming broadcast (group address) telegrams are put in the TelegramQueue.
For point to point telegrams we have a special package `management` where telegrams addressed to xknx' individual address are handled. 

Have a look at 
https://github.com/XKNX/xknx/blob/main/xknx/management/procedures.py

This module currently only handles outbound connections, but it should be possible to handle inbound as well. 
We'd love PRs adding more procedures to that module 😉

knxjoerg, 2022-09-24:
Hi Matthias,

thank you for your quick response.

I have spent two nights in order to figure out, how XKNX works internally
and how to make management telegrams work. XKNX is an impressive work of
software that exceeds my programming skills by far. So let me exress my
dire respect - and sorry, I cannot write any adequate contribution.

Is there any way, to simply tell XNNX to trasmit binary telegrams bytes
like B0 11 FD 04 06 60 80 and pass any received KNX telegram binary to a
callback function without further checks?

Best regards, Jörg.

farmio, 2022-09-24:
Maybe the best way to start is to describe what exactly you want to achieve.

> Is there any way, to simply tell XNNX to trasmit binary telegrams bytes

not really. Everything is built on either the `Telegram` or the `KNXIPFrame` abstractions. Sure, you can hack your way around that by modifying eg. the Tunnel class, but I'd not recommend doing that.

> like B0 11 FD 04 06 60 80

👀📎💬 looks like you want to do a T_Connect using an extended frame format. 
Extended frame format is not explicitly supported by xknx - I have no idea if it would handle it silently or just raise some error.

> pass any received KNX telegram binary to a callback function without further checks

again, no, there is no build-in callback supporting that directly. Any received data is immediately parsed to a `KNXIPFrame` before passing to any callback.

<hr>

To do a outgoing peer to peer connection you can simply do

```py
# create a transport layer connection context - it sends `T_Connect` on call and `T_Disconnect` when exiting
async with xknx.management.connection(address=IndividualAddress("1.2.3")) as connection:
    # the `request` method sends a request and returns the response
    response = await connection.request(
        # this is your request payload
        payload=apci.DeviceDescriptorRead(descriptor=0),
        # and this is the response you expect - if it times out or receives a different response it will raise an exception
        expected=apci.DeviceDescriptorResponse,
    )
    # do something with `response` or send additional requests...
```
For an example on how to handle exceptions for that, again have a look at the `xknx.management.procedures` module.

Incoming management connections are not supported right now. See https://github.com/XKNX/xknx/blob/1d53381da7a0ad061b76f63f11e1fd13219a1a78/xknx/management/management.py#L56

knxjoerg, 2022-09-25:
Let me describe what I want to achieve. In my private KNX installation I am
using an increasing number of self-built KNX devices, which are not
comercially available like e.g. a generic KNX servo motor controller. They
are all based on AVR microcontrollers. I really would like to do in-system
software updates for those microcontrollers over KNX. So basically I am
talking about a python-based KNX-AVR programmer. This requires a bunch of
outgoing and incoming management telegrams.

It is goot to know, how outgoing management telegrams can be transmitted
over XKNX. But as long as I cannot properly receive the incoming management
telegrams yet, there seems to be no way to establish a point-to-point
connection over a KNX IP Tunnel via XKNX. So I will probably use a simper
KNX access method instead like e.g. a TPUART2 over serial UART, where
"simple" refers to a hardware-developer perspective.

Many thanks, Joerg.

farmio, 2022-09-25:
Not sure I can follow. 

What do you need incoming management connections for in that case? Is xknx running on these controllers?

Incoming management telegrams that are responses to outgoing requests can be received just fine with the example above. I don't think a device should send p2p telegrams without a previous request, should it?

knxjoerg, 2022-09-27:
the code snipped looks very promising.

> Not sure I can follow.

Exactly. For me it feels pretty much the same. Guess, we share a
communication problem rather than a problem with the software itself.  If
you like, we can have a video discussion. -?

So I still do not understand how I can carry out all the steps required for
a programming process with XKNX, i.e. steps like:
* identify a device with programming button pressed or
* wait for a programming button pressed on a device to be programmed
* make the connection to the specific device
* send configuration data like e.g. write a new physical address or
* write the group communication table
* restart the device
* close the connection

I honestly have to admit, that I also do not understand the GroupAdress
communication completely. For instance a simple operation like
telegram = Telegram(source_address="0.4.17", destination_address="0/4/8",
telegramtype=TelegramType.GROUP_WRITE, payload=True)
xknx.telegrams.put(telegram)
raises a lot of type errors. Maybe an example in the documentation would be
helpful.

farmio, 2022-09-27:
> * identify a device with programming button pressed or
> * wait for a programming button pressed on a device to be programmed

These two are just not implemented yet. `03_05_02 Management Procedures v01.09.02 AS.pdf` describes the steps to be done.

> * make the connection to the specific device
> * close the connection

This is already covered by my second post here.
```py
async with xknx.management.connection(address=IndividualAddress("1.2.3")) as connection:
```

> * send configuration data like e.g. write a new physical address or
> * write the group communication table

Again not implemented directly - as in, there is no function to pass a current and new IA and that's it. This would be an additional procedure function.
I guess it can be done manually by sending the right requests in the right order.
```py
    response = await connection.request( ... )
```

> * restart the device

This is implemented as a procedure (DM_Restart): https://github.com/XKNX/xknx/blob/01488c42f0c169871c1f429300dd5f75f5763111/xknx/management/procedures.py#L17
Here is an example on how to use it exactly:
https://github.com/XKNX/xknx/blob/main/examples/example_restart.py
As part of a bigger procedure it would have to be called manually in the existing connection context. See implementation on how to do that.

> telegram = Telegram(source_address="0.4.17", destination_address="0/4/8",
telegramtype=TelegramType.GROUP_WRITE, payload=True)

Please just have a look at the signature of `Telegram.__init__()`: 
https://github.com/XKNX/xknx/blob/main/xknx/telegram/telegram.py#L33-L45
There is no `telegramtype` keyword argument and I don't know where you get `TelegramType.GROUP_WRITE` from.
Consider using a proper IDE with helpers like code completion etc. (eg. VSCode, PyCharm) this helps you find such errors easily.
Easiest way to send telegrams (high-level) is described here: https://github.com/XKNX/xknx/blob/main/examples/example_send_telegrams.py

> Maybe an example in the documentation would be
helpful.

I fully agree that our documentation is not very extensive and could be much more. 
The things you try to use are considered low-level - they are quite new (management) and may be subject of change in near future (not extensively tested). We try to write readable code, have good docstrings, extensive comments and type annotations to help new users understand how these low-level things work.
Some examples can be found in the `examples` directory.

> Guess, we share a communication problem rather than a problem with the software itself.  If you like, we can have a video discussion.

Feel free to join our Discord server (invite link can be found in the Readme) for a more direct conversation.

closed 2023-01-02 by github-actions[bot]